prusa3d/Prusa-Firmware #704: Feature Request: linearity correction Gcode

kwaad2, 2018-04-30:
@stahlfabrik has been messing with the linearity correction and it appears that it will fix the Moire pattern many of us have.

I was curious if it would be possible for it to be changed by Gcode, so that you could essentially do a "linearity correction tower" to test the linearity correction, rather than having to *literally* print 20+ cubes.

One...
At...
A...
Time...

stahlfabrik, 2018-04-30:
Haha, you only have to print so many cubes if you do not know what you are doing  -> That is criticizing that there is literally no documentation of what the new setting does and how to decide what to aim for. So I took @XPila calibration gcode that he published and startet wit 1.110 and then went down in 0.010 steps to 1.030. I also went higher than 1.110 to 1.130. At 1.130 the moiree was very visible. At 1.060 it started to disappear, so I decided on 1.055. What I also do not know yet: on lower values the moiree seems to be gone, too: So could one just take the lowest value? Or is there a benefit of finding the sweet spot where the moiree disappears

I really did not know from the start what to watch out for. If there was documentation it would be faster. A series of some hires photographs on what to look out for would be a huge help-

It would be nice if the printer would print some cubes one after the other on the bed, each with different settings. Then you could decide on the value - maybe something like this COULD be implemented? I have no idea.

kwaad2, 2018-05-02:
@stahlfabrik Yeah, no I agree, *I* could do it much more "coarsely" however I think it would be easier for *ANYONE* if you could just do a "linear tower".

You could literally do a different correction every 1mm.

1.030 - 1.130 = 100 steps.
Run it by 0.001 per 1mm of the "tower".
And run the tower 100mm tall.

Essentially in THEORY, you will be able to watch the Moire effect change all the way across the tower, and "measure" it with a pair of calipers. Boom, single ~1-2hr print, to get it dialed in.

Regardless of how you say to dial it in. Even if you "roughed" it with 0.03 per first prints, you would still need 3-4 prints, to get an idea of where you need to be, then from there, another 3-4 to "dial it in". That's 6-8 prints 10 minute, using advanced logic, that many people struggle with. That's going to take around 2-3 hours, of ACTIVE tuning on the printer. vs, a 1-2 hour "calibration print". That you simply "start", "Remove" "measure" "Change Setting" "Done".

To explain easier is to say instructions.
1. Print a test cube at 1.100
2. Print a test cube at 1.130
3. Print a test cube at 1.070
4. Print a test cube at 1.040
5. Go get on the forums, and ask someone who knows what they are looking for, to tell you what your next set of prints will be. (Many casual users will NOT be able to do this properly, so to simplify things, get someone who knows what they are doing to do it)
***STEPS FOR FORUM MEMBER***
6. Tell the user to print another 5 test cubes at 0.005 incriments between the two best looking cubes.
7. Tell the user to return to the forums, and ask them to print another 4 prints in 0.001 incriments, and once again respond the best answer.
8. Tell the user to set that best answer on their printer.
(This has resulted in 13 individual prints, that will take around 10-15 minutes of the users time. EACH, resulting in around 2-3 hours of "work" for the casual user)

Using a gcode tower.
1. Tell user to print linear Gcode tower.
2. Tell user to find the area that looks the best to him.
3. Tell the user to measure from the bottom to there with a caliper. (or ruler)
4. Take whatever mm measurement you have, and add it to 1.030. Example: 46mm was optimal. 1.030+0.046=1.076.
5. Tell the user to set that number as their linear correction on the printer.

This way, the user prints a single ~1-2hour print, that will print "unattended". And the only "time" it takes the user to perform this "calibration" is, downloading the gcode, starting the print, measuring, and setting the linearity. It might take 20 minutes of their "active" time, from start, to finish.

mvasilakis, 2018-05-11:
@kwaad2 So we just need to put TMC_SET_WAVE_E40 into the start g-code and that will set a correction?

kwaad2, 2018-05-11:
Yes.

For example
M911 E26     ; This is a line of gcode.
TMC_SET_WAVE_E40    ; This is also a line of gcode. This sets Linearity correction to 1.040
TMC_SET_WAVE_E140   ; This sets Lineraity correction to 1.140

They updated the wiki for setting correction.
https://github.com/prusa3d/Prusa-Firmware/wiki/Extruder-linearity-correction-calibration

I thought to myself. "They must be changing it in the gcode somehow".
So I did a little reading of the gcode in their calibration tower.

RetireeJay46, 2018-07-13:
I printed the test tower, and I couldn't really see any difference between top and bottom.
It is a bit confusing what they are doing on the tower.  The last adjustment (at the top) uses this code:
TMC_SET_STEP_E1046
;104.8
But the web site [https://github.com/prusa3d/Prusa-Firmware/wiki/Extruder-linearity-correction-calibration](url)
seems to say that the top of the tower is at a correction value of 1.20.  I don't see the correlation between 1.1048 and 1.20!

kwaad2, 2018-07-13:
I'm not sure what TMC_SET_STEP does specifically... it simply changes the "phase" of the step... somehow.... I'm not sure how to explain it, or even really understand how/why/what it does myself.

You are looking for TMC_SET_WAVE. That is what actually changes the linear correction.

RetireeJay46, 2018-07-14:
Thanks, kwaad2.  I'm still not sure I've made any improvement, but the basketweave was never a big deal for me anyway.

Elocai, 2019-04-23:
does this relate to some tmc parameter or do you know whats that called in the manual? I´m on klipper and have the same moire

Lichtjaeger, 2019-07-15:
Has anyone finished the calibration tower files for X, Y, and Z yet?

I have done the E calibration, but there is no place with no lines. So I tried to pick a place where they're least visible:

[image]

agryson, 2019-07-23:
> Has anyone finished the calibration tower files for X, Y, and Z yet?

I was just about to try doing this (I am lucky - no moire effect but vertical linear banding in Y)

Am I correct in assuming doing an X Y or Z tower is as simple as changing the gcode to 
`TMC_SET_WAVE_X0` ,  `TMC_SET_WAVE_X30` etc. instead of `TMC_SET_WAVE_E0`?

changedsoul, 2020-12-13:
What does Linearity Calibration mean exactly?
Is it similar to E Steps calibration, in the sense that what is being adjusted here is how much the extruder moves per programmed distance?
For example, TMC_SET_WAVE_E0 = it moves by a factor of 1, and TMC_SET_WAVE_E40 = it moves by a factor of 1.04 per programmed distance?
Or is it adjusting something else?

closed 2023-09-05 by github-actions[bot]