prusa3d/Prusa-Firmware #707: 3.2.0-RC2 - MK2.5 XYZ Calibration fails

albegouws, 2018-05-01:
When running XYZ calibration (was last run on 3.1.3.245) I get an error calibration point not found on the first point.  Without changing anything(belts etc) just rolling back to 3.1.3 it completes the test with no issues. Tested a few times, fails on 3.2.0 RC2 every time.

PavelSindler, 2018-05-02:
Could you please send serial log from failing xyz calibration?

albegouws, 2018-05-02:
I have run a few M45's via pronterface.  You'll notice on the first one I loaded the wrong firmware (RC1).  Disconnected a few times.  The failures all failed on the second point. Last one ran through successfully.
[output.zip](https://github.com/prusa3d/Prusa-Firmware/files/1968351/output.zip)

b3niup, 2018-05-05:
Same issue is described here:  #616

nbun, 2018-05-16:
I just encountered the same problem while trying to do a re-calibration. The calibration that's part of the wizard worked initially when I upgraded to 3.2.0, but the xyz-calibration from the setup menu failed 4 times. When I ran the wizard, it worked again. Maybe this has something to do with the heating process before the calibration or the temperature calibration of the probe?

closed 2023-09-04 by github-actions[bot]