prusa3d/Prusa-Firmware #708: Strange extruder behaviour after 3.2.0 upgrade

Paulbsh, 2018-05-01:
After the 3.2.0 upgrade I was about to print the same model that I already printed about 25 times with 3.1.3 
When it prints the first layer, I could observe that the extruder puts down blobs at certain spots (Marked green in the pic) with a noticable click (extruder gears speed up for a fraction of a second). The click also appears on the turning points of the first layer (marked red) in a rising manner. 
Just to be clear: this was NOT the case with 3.1.3 with the exact same gcode.

Is this a known issue?
[image]
Video of the issue:
https://www.youtube.com/watch?v=-5NeWJXItd0&feature=youtu.be

alientek, 2018-05-01:
Check your belts

Paulbsh, 2018-05-01:
Dude, seriously?
Did you even read what I wrote and watched the video?
The issue was not there in 3.1.3 (printed the same model 5 minutes prior to the upgrade).
It's an extruder issue, not a belt issue

alientek, 2018-05-01:
@Paulbsh Im having no issue's and it sounds like a screw is loose on the x,y motors.

ericclinedinst, 2018-05-01:
@Paulbsh this alientek guy is trolling the github posts. He's ober on 602 and many others doing the same. I think he's a Prusa employee. Or King Prusa himself.

Paulbsh, 2018-05-01:
I see. Maybe someone without a complex has a clue?
I repeat: This does not happen with 3.1.3

moggieuk, 2018-05-01:
I was also thinking that your extruder was missing steps as a result of reduced current.  You can push the current back up with gcode to test.  Alternatively slow your print down (reducing flow rate) as a test of the theory.

Paulbsh, 2018-05-01:
@moggieuk, wouldnt that mean that the gears would jump backwards?
They're jumping forward at the mentioned points

moggieuk, 2018-05-01:
Hmmm, I just watched your video, very strange. It's hard to tell what the extruder is doing but it certainly looks like it is trying to lay down a blob at the end of every line.  Thinking of what has changed in the firmware other than extruder current the other thing is linear advance.  Does your gcode have a silly k-factor setting that would have been ignored in v3.1.3?

Crunch69, 2018-05-02:
i had a very similar issue and it was caused by something in my gcode script.  I did not do any heavy analyses as I needed to get a print started I just replaced my script I used in 3.1.3 with the simple 1 below and it went away. 

G28 W ; home all without mesh bed level
G0 X50 Y50 Z0.15 ; this is a good PINDA heating position
M104 S215; set extruder temp
M140 S60; set bed temp
M190 S60 ; wait for bed temp
M109 S215; wait for extruder temp
M860 S35 ; wait until PINDA is >= 35C
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E

PavelSindler, 2018-05-02:
Hello @Paulbsh , can you send the gcode file?

wavexx, 2020-01-27:
I think I fixed this without knowing with #2416 which didn't handle G92 E0 correctly, likely causing stuff like this.

Prusa-Support, 2022-05-18:
This thread could be related to issue #765, or could be solved by PR #2416 mentioned above.
Also, this thread has been silent for years.
For these reasons, this issue will be closed.

Please check the issue and the pull request mentioned above to learn more about the conclusion of this case. 
Comment below or feel free to open a new issue if you feel this issue deserves a separate discussion.

Michele Moramarco
Prusa Research

closed 2022-05-18 by Prusa-Support