prusa3d/Prusa-Firmware #720: FW 3.2.0-RC2 If you pause an SD print printer will not resume the paused print

Crunch69, 2018-05-06:
FW 3.2.0-RC2 If you pause an SD print and the select resume the printer will not resume the print and extruder and heat bed shut down.  In order to print you must power cycle printer to print again.

GrafiTom, 2018-05-10:
Yep. Happened here to. When I hit resume it displayed "Heating......" but the system was cooling down. I then dialled preheat PLA and when it hit the threshold started to continue the print.

AikiAviator, 2018-05-12:
Akin to issue #734.

Prusa-Support, 2022-05-23:
Hello! Sorry for intervening so late in this thread. I can confirm the issue has been fixed in the meantime, closing this thread based on this info. In case anyone might be experiencing similar issues, do not hesitate to let me know info@prusa3d.com mentioning me in the subject ("attn: Alessandro"). 

Alessandro Pantaleo
Prusa Research

closed 2022-05-23 by Prusa-Support