prusa3d/Prusa-Firmware #721: Higher z in home position would make it easier when loading filament

hanpal, 2018-05-06:
A. The home position is very low in z and it's hard to see if the loading is successful without bending down and directing a lamp under the nozzle. Also note that all users are not 20 years old with perfect sight.

B. Sometimes the filament will also pile up and clog the nozzle, as a result of the nozzle being so near the heat bed.

It's tedious to adjust z manually before loading filament but this is what I do now and then to solve both problems.

A simple solution for both problems would be to have a higher z value in the home position.
No extra memory in firmware required...

stahlfabrik, 2018-05-06:
Hi @hanpal 

I do not know if your request is possible because "home" means "0" - where the ends are.

It would be nice though if with "autoload filament" Z would raise - but that would require that the printer is homed:-) So it is unrealistic too

Do you know that long pressing the dial knob takes you quickly to "move z". So changing z is not tedious at all. I think that might help you getting along.

hanpal, 2018-05-06:
OK didn't know about that long pressing. "home" cannot be zero in z since the nozzle is some distance above the heat bed already. I just want it to be 50-100mm higher to simplify loading and cleaning. Should not be a big deal and are there really any big disadvantages?

stahlfabrik, 2018-05-06:
I am not sure - there  might be pros and cons for both. 

I just think that loading filament can also happen without the printer being homed. So why change home's z hight if you do not even know where z is:-) For me personally the long press is good enough for that aspect (loading filament)

nospoon, 2018-05-07:
I'd say having the print head move to the top after finishing the print would also help with decoupling the steel sheet. I always find myself manually moving the Z up with smaller prints just to get the extruder out of the way.

koenkooi, 2018-05-08:
I've changed the end gcode in slic3r to always raise z to 209, it's a huge time and motivation saver :)

rommer, 2018-05-08:
Raising the Z height will effect mesh bed leveling since the pinda probe will not be over 35C which is needed to get a decent first layer. 

The long press works very well for me as well.

stahlfabrik, 2018-05-08:
The solution is really to discover and use the end gcode imho.

All the way up is the worst I can imagine. That noisy Z axis should run as little as needed:-) And as @rommer said. The PINDA likes it warm so it likes to stay low.

No doubt could Z raise to a minimum value after a print so that the heatbed sheet could be easily detached - but again: the end gcode is your friend

sniper85500, 2018-05-22:
@hanpal 

so why you just change the prusa firmware just for you to get a higher Z? I dont want it on my printer :-D

nospoon, 2018-05-22:
I don't see how rising Z after finishing the print can possibly affect bed leveling which is done at the begining. You would still need to wait for the bed to cool down before taking the part off and starting a new print, so I really see no relevance here.

bubnikv, 2018-05-22:
With the upcoming Slic3r 1.40-beta we will move the print head couple of cm
up at the end of the print.

hanpal, 2018-05-22:
You only need to raise z if below a certain level after finished printiing, 5cm or similar. If lower, raise to this z.

bubnikv, 2018-05-22:
@hanpal 
> You only need to raise z if below a certain level after finished printiing, 5cm or similar. If lower, raise to this z.

This will be freely configurable through a Slic3r custom end G-code.

closed 2023-09-04 by github-actions[bot]