prusa3d/Prusa-Firmware #725: [MK3 - 3.2.0] After finishing every layer, extruder moves to beginning of X-Axis.

jl1990, 2018-05-09:
As title says, after one layer is finished, extruder moves to the beginning of the X-Axis. After reaching that point, it comes back and continues printing again...
Please, fix soon, as it increases a lot printing time and leads to other printing problems.

Thanks

GurliGebis, 2018-05-09:
Are you using Octoprint? If so, do you have the timelaps feature enabled?

jl1990, 2018-05-09:
Yes to both questions... Maybe I got some configuration wrong?

stahlfabrik, 2018-05-09:
No, that is what octolapse does. You could close this issue!

jl1990, 2018-05-09:
Yes, exactly I had octolapse installed, I will read about how to configure it properly. It's strange that this started happening to me from RC2 in advance when I had it installed earlier. Thanks!

closed 2018-05-09 by jl1990

stahlfabrik, 2018-05-09:
Yes the coincidence is funny:-)