prusa3d/Prusa-Firmware #728: Extruder crashing on 3.2

morimorida, 2018-05-10:
So I flashed a new firmware 3.2 and did calibration (without any issues during calibration).  Started to print a 20x20mm cube and encountered crash toward last 20 layers or so.  In total, for each cube, extruder crashed 5-7 times and after it crashes, it resumes at where it crashed BUT it does not print for 30 second or so and hovers over the printed object before resuming.  I tried again and same thing happened at practically same location.  Please see below folder I created for pictures and video.

https://drive.google.com/drive/u/0/folders/12MYPnS3ruTtrXxSVnYjN3_ujLCWOKB02

1) Extruder crashes, hovers over it crashed for least 30sec before resuming the print.
2) It shifted the layers.
3) top portion has a hole even if I have 5 top layers on slicer setting.
4) Maybe my imagination but my webcam streaming was stuttering (which is connected to Raspberry Pi3 and Repetier sever).
5) 

My environment:
1) Raspberry Pi3 with Repetier server connecting the printer and webcam with USB cable.
2) Sliced with Simplify 3D. 1.0mm height, 205C nozzle, 60bed, extrusion width 0.45, nozzle diameter .40mm, extrusion multiplier 1.00m, 20% infill, 5 top, 4 bottom layer with 2 perimeter, printed at 100mm/s

I am printing the same cube using the SD card to isolate if it's a communication issue.  I'll let you know once completed.

Oh one more slight annoyance but repetier server says 59.8C on bed and printer says 60C.  This discrepancy didn't happen with prior firmware.  However, prior firmware did have a problem where after the print was finished the bed temperature indication on repetier server does not go back down to room temperature.  On MK2S that I have do not have this problem.  Thought I mention this additional problem.

PavelSindler, 2018-05-10:
We are testing your gcode on multiple MK3 printers now. In case that crash detection is causing layer shifts, re-running selftest should help (it contains accurate rehoming calibration).
I don't understand what happened when printing from repetier server with disabled crash detection. What do you mean by extruder temperature 50C? You mean extruder stepper temperature or nozzle temperature? When it was measured?

PavelSindler, 2018-05-10:
Yes, extruder motor temperature is quite high. If extruder stepper temperature is too high, filament can become slightly soft, which causes filament deformation and can end with clog. We know about it and that's why we lowered E current in 3.2.0.

morimorida, 2018-05-10:
Thanks for the explanation.  If you want me to test anything further, please let me know.

morimorida, 2018-05-11:
I'll do test when I get back home today

1) self test and reprint from repetier server with crash detection off.
2) disable LA and print.
3) check extruder motor temperature start and end of each print.

morimorida, 2018-05-11:
Today's test:
4) Left: Crash detection Off, No LA
5) Right: Crash detection On, No LA

https://drive.google.com/drive/u/0/folders/1yuMoSSLzM8u79_dq710sDNBXXRkv9dNq

Both cube did not have layer shift. No holes on top layers (I think this is since there was no layer shift).  When crash detection was on, nozzle crashed once, went home, came back to where it crashed...THEN it hovered over the printed object for 30 sec or so then started to print again. Extruder temperature measurement right after the prints were 42C. Also, was using repetier server+usb cable to print instead of using SD card.

Other notable change on slicer I did when I disabled the slicer was added back coasting (0.2mm) and wipe nozzle (2.00mm).

Hope this test helps...

rommer, 2018-05-11:
I thought there was some kind of issue using repetier server? It was a few versions of the firmware ago and I don't recall ever seeing a fix for that issue. Might have missed it though.

morimorida, 2018-05-11:
Nah, that Repetier problem was from 3.1.0 that was fixed along with Octoprint problem on 3.1.3. 

So right before I slept yesterday, printed the egg and it clogged again 2/3 into the print.

I think I had similar issue before on my MK2 when I upgraded the fan to Noctua fan.  I just bought same 40x40 mm fan that I purchased for my MK2 on Amazon and I'll swap the fan and print again to see if problem still persists.

So far without LA, I don't get layer shift.  Clog still happens but I think this is mainly when printing a small object.

morimorida, 2018-05-22:
I changed the Noctua fan with 10cfm fan to see if cooling the heat sink more would avoid the clogging but this did not help.  The tip of the egg did not print.  So I think it's bondtech gear area that's softening the PLA filament causing filament feeding issue.

As for Repetier server problem, I think this problem is same what we are having with Octoprint problem.  I feel this Repetier server problem will be resolved with resolution we have for Octoprint.

Prusa-Support, 2022-06-08:
This thread has been silent for years. Moreover, some legacy FW and design upgrades may have improved some concerns aspects by now. Therefore this issue will be closed for now.

The layer-shifts and the crashes can happen for several reasons. 
https://help.prusa3d.com/en/article/layer-shifting_2020
Poor print quality can also be a cause for portions of the print to lift up, generating an obstacle for the X and Y movements.
In this case, the print parameters may actually deserve more investigation too.

As per the concern of the extruder motor being hot, please check the conclusion of issues [765 ](https://github.com/prusa3d/Prusa-Firmware/issues/765#issuecomment-1120239675)and [1757](https://github.com/prusa3d/Prusa-Firmware/issues/1757#issuecomment-1133245662).
Poor print quality may also be a consequence of the concerning aspects listed in the final comment on these issues.

This issue will be closed. Please comment below or open a new issue if you think the issue deserves further study.
Also, please consider contacting the Technical Support before posting about new similar problems.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research