prusa3d/Prusa-Firmware #729: 3.2.0 Firmware MK3 - Extruder skips during initial purge and during 1st skirt loop

Petclaud, 2018-05-10:
The new firmware fixed the extruder skipping as earlier reported by me in [https://github.com/prusa3d/Prusa-Firmware/issues/696](url).

Great, thank you very much Prusa! However I do experience still a single 'click' (extruder skipping) during the initial purge line at the point where the extruder speeds up the flow. Also a click can be heard during the 1st part of the extrusion of the outer skirt/loop with results that the initial loop is not fully closed. It looks like the starting speed - or alike - is too fast. I have experienced this already several times. Both with PETG and with PLA

morimorida, 2018-05-10:
> I think it has something to do with Lin Adv, try to disable it.

I'll do test when I get back home today

1) self test and reprint from repetier server with crash detection off.
2) disable LA and print.
3) check extruder motor temperature start and end of each print.

Zawadaki, 2018-05-10:
Extruder skips during initial purge and during 1st skirt/loop

I just upgraded firmware, made calibration end everything is OK besides of skipping extruder

extruder is cold
selftest OK
I dont disable LA because it's not the point to disable most significant feature, if I disable LA i can also flash 3.1.3 firmware and made this update useless

bring back old currents for extruder !!!

alientek, 2018-05-10:
@Zawadaki It did it with old currents too.

Anyways its just 1 skip during purge so what? it has nothing to do with the actual print so it doesn't matter.

Zawadaki, 2018-05-10:
Now it's skipes during normal print and leave marks on surface !!!! it matters

alientek, 2018-05-10:
@Zawadaki Sounds like a you issue, not firmware.

Zawadaki, 2018-05-10:
yea skipping during test print using prusa filament, prusa g-code, prusa printer and it's my fault, great maybe earthquakes are also my fault

rommer, 2018-05-11:
@Zawadaki, someone is trying to help you and if you don't like what they have to say then it would be best if you ignore them. FWIW, I've not experienced the same problem as you.

In case you are not aware, the extruder current was raised after rc2. It's still lower than what it was before but damn the extruder temperature is MUCH cooler. I believe this will help a lot of people who were having filament issues in the bondtech gear area.

The only time I've ever had an issue with a skip on the prime line was if my live z was really low.

bubnikv, 2018-05-11:
The extruder skipping at the purge line may be "normal" with linear advance
as there are extreme changes in exrusion rate and the LA algorithm produces
strong sudden jerks in the E axis. We may just disable LA for the purge
line in the Slicer profile.

The LA on MK2 works a bit differently than on the current MK3 firmware: the
MK2 method may create sudden jumps in the XY plane if the extrusion rate
changes rapidly, while the MK3 method tries to keep the XY movement smooth,
but then it has to jerk the E axis quicker. You may therefore see some rare
extruder skips at the start of a rapid extrusion, that is mostly at the
beginning of a wide gap fill. Such extruder clicks will likely lead "just"
to a bit of an underextrusion of these moves. The correct solution is to
vary the extruder pressure smoothly in Slic3r, which I tried to prototype
about a year and half ago (it is the volumetric extrusion rate
positive/negative limit), but I never had time to finish. If just the day
had some more hours.

Also if you are getting too many extruder skips, please make sure your
bondtech extruder tension is correct. Too much tension is bad as it causes
too much deformation therefore drag. Counter intuitively it is often too
much tension on the bondtech extruder, which causes the E skipping.

Zawadaki, 2018-05-11:
I lowered the tension as instruction describe and is still skipping during normal print so what I shel to do next to fix this problem ?

Petclaud, 2018-05-11:
Just to add my 2 cents to the discussion: I have printed now several hours and several prints with the new firmware (happened to be a couple of days off) and in my case, only a single skip during the initial purge and occasionally during the 1st skirt/loop. This initial loop is then not closed and sometimes also it curled up with the possibility to be 'caught' by the extruder in the next round. 

I have NOT experienced any skips later in the print (at least not that I have heard) and the prints look very good. So indeed it is a minor issue for me. Just wanted to let you know about my observation and if this could be avoided through firmware it would be another improvement. 

Thx Prusa team. Keep it up!

Zawadaki, 2018-05-11:
no one asks you for a sentence

alientek, 2018-05-11:
@Zawadaki Take it easy big boy, if no one else is complaining about it dont you think you have done something wrong?

Zawadaki, 2018-05-11:
this fucking printer can't even print spirale vase right, so maybe some advice because I received only one so far ?

mvasilakis, 2018-05-11:
If I'm not mistaken the original current is E36. M911 E36 will set the extruder to the original current. You can try that and see if it solves it for you. Also maybe play with detraction speeds. (New E is 26, For heat mitigation I'd find the value between the two that solves the issue.)

I believe the full original current:
M911 X14 Y20 Z30 E36
New:
Holding currents: M911 X16 Y20 Z35 E26
Running currents: M912 X16 Y20 Z35 E26

Brainlazy, 2018-05-11:
Doesn't Marlin's current version of Linear Advance already address this?
Was there some benefit in using an older implementation that doesn't automatically conform to jerk limits?

mvasilakis, 2018-05-11:
@Brainlazy I'm spending allot of time perusing forums, git bugs and groups looking to see what others are doing to solve their issues so that maybe I can improve my own. From what I've seen there is no one setting for every printer. 

While a majority of users may not have overheating issues with the old settings enough did that Prusa dialed down the currents. That move brought other problems to surface such as a few where lower currents cause problems where there were none before. 

Since the motors, PSU's, boards, wires etc.. are built to meet a spec range some will fall in the higher ranges and some in the lower. This deviation inevitably yields statistical outliers that will require tweaked settings. 

OP stated he's getting clicking throughout the print. This is most likely a tension issue. OP stated that he's messed with the gear tension to no avail and requested code to revert current. I simply obliged. Hopefully he updates us. I'm curious to see if increasing current fixes his issue.

Petclaud, 2018-05-11:
@Zawadaki ; interesting that you write 

> no one asks you for a sentence 

I think I have opened this thread! Anyway I hope it will be resolved for you. For me the skio that I experience in the initial purge and the occasional skip in the 1st loop is not a major matter. Skipping during printing of course will be an issue - as you experience.

rommer, 2018-05-12:
@Zawadaki do you really think anyone wants to lift a finger to help people with such a piss poor attitude? 

Come to think of it, you probably do. So sad...

rommer, 2018-05-12:
@mvasilakis, the current was lowered to E26 in the alpha but was raised to E30 in the released version.

Arksine, 2018-05-12:
I noticed a skipped step during the purge and at the start of the skirt when printing with PETG.  Didn't notice it with PLA but I haven't used much of it on 3.2.0.  I haven't noticed any skipped steps during an actual print.

I think disabling LA during the Purge is probably the correct solution, atleast in the short term.

Brainlazy, 2018-05-12:
I was under the impression that LA is "disabled" as long as the K value is 0 (which should be the firmware default) and it would only become active when M900 defines K in your gcode.  Why then is the extruder skipping during the purge line if K isn't being set til just afterwards?

Panayiotis-git, 2018-05-13:
@Brainlazy And the second print will start with the K value of the previous print.

Brainlazy, 2018-05-13:
Add a M900 K0 to the top of the slicer's Start gcode then?

rommer, 2018-05-13:
I'd add the M900 K0 to the end code so in the event you print something from the SD card or an older gcode file you don't start the print with the wrong LA setting. It's probably better to print with no LA then have the incorrect LA Mcode.

Brainlazy, 2018-05-13:
Yeah, probably best to clear the K value at both the start and end of the gcode. Might avoid skipping the step in the prime line and would keep things clean for outside gcode that might not have a defined K.

Panayiotis-git, 2018-05-13:
For this reason, I always do a reset before start a new print.

vestom, 2018-05-29:
FWIW, I am also experiencing routinely extruder skipping during the purge line since updating to 3.2.1. I have never seen it before. It seems likely it is due to "overdriving" the extruder due to the gain of Linear Advance during the extreme change in extrusion during purge. 

Instead of just "fixing" it by disabling Linear Advance, maybe it would be better to implement a gain saturation of Linear Advance to avoid the problem in other inopportune situations?

closed 2018-07-25 by Petclaud