prusa3d/Prusa-Firmware #730: [Question] Best way for compiling firmware?

stelgenhof, 2018-05-10:
My apologies as this is not so much of an issue but rather a question.

When compiling the firmware for the different variants (MK2S, MK2.5, MK3), what is the best way to do this? I am a bit confused by the README's in each branch.

For example, if I like to compile for the MK2S should I checkout the MK2S branch and following the enclosed README instructions? Or, can I use the MK3 branch to compile for the MK2S since it has the MK2S variant included?

Cheers! Sacha

FischOderAal, 2018-06-05:
It is quite simple really. The firmware works for all variants, which functions your printer has is defined in the file "_configuration_prusa.h_" (see step 4/5). The instructions in the readme are quite good.
1) install Arduino IDE (preferably 1.6.8)
2) add the Ultimachine board manager
3) download the complete package
4) go into subfolder "variants" and take the one you need (i.e. "_1_75mm_MK2-RAMBo13a-E3Dv6full.h_" for MK2(S)) and rename it to "_configuration_prusa.h_"
5) copy the file to the folder to the parent directory
6) open "_firmware.ino_" in the Arduino IDE (it should automatically open all other necessary files)

Bob's your uncle. I don't know if you may have to install some libraries, but it can be done via the manager inside the IDE.

Maybe this helps (I'm assuming you're German?!):
https://drucktipps3d.de/firmware-selber-kompilieren-wie-fang-ich-das-an/
(you do not need to change the Library LiquidCrystal any more)

stelgenhof, 2018-06-05:
@FischOderAal Thanks! Perhaps indeed my question needs to be rephrased :)

My question is not so much about _how_ to compile the firmware, rather how the repository is organized. There is a MK2S branch, but also the MK3 branch contains a variant for the MK2S.

However, from a different conversation I understand that the goal is to unify the source code for all models. Which is currently in process to my understanding.

BTW: I'm not German, but Dutch :) I know my first name is quite common in Germany.

Cheers! Sacha

gorillamotors, 2018-10-11:
FischOderAal
I am assuming that when you compile the the firmware the software uses the file configuration_prusa.h and the looks at all the h files in the directory for all the specs called for in the configuration_prusa.h file and then creates the compiled file. Right?

Jim

trevjonez, 2018-10-11:
given that this project must be built with older and specific versions of arduino/gcc it is crying for some docker images to help tame the environment management side of things. 

also first party clion project support would be a nice to have :) 

looks like it should be possible. https://github.com/search?q=arduino+docker

FischOderAal, 2018-10-11:
@gorillamotors you should use the source code zip file. It has all the files for all the variants but in the end you just choose the correct configuration file. Be sure to use the correct version of Arduino and NEVER update it.

hoppke, 2018-10-11:
A possibly easier way to build the firmware is to forget about Arduino IDE and go with platformio? Install platformio, check out the firmware, then (for MK3) in the firmware dir run:
`pio init -b reprap_rambo`
then edit platform.ini so it looks like this:

```
[env:reprap_rambo]
platform = atmelavr
board = reprap_rambo
framework = arduino
build_flags = -w -Os -Wl,-u,vfprintf -lprintf_flt -lm -Wl,--gc-sections

[platformio]
src_dir = Firmware
build_dir = out
```

Activate the setting variant appropriate for MK3:
`cp ./Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h ./Firmware/Configuration_prusa.h`
(or just symlink it I suppose?)

And then it's just `pio run` to compile/produce the .hex file. Platformio deals with downloading the right compilers and libraries...

Then there's the multi-language stuff to solve, but I guess the only real issue there is that the Prusa scripts weren't meant for use by non-Prusa folks, and that they're Windows(?)-centric (so hopefully no hard problems to solve there, it's just that someone would have to make the build a bit more generic)

One benefit of platformio is that it doesn't interfere with whatever arduino ide you may (or may not) have installed, and the platformio.ini file could be checked in, so you would get the compiler flags set the way they should be straight in the repository, it apparently lets you control the versions of toolchains needed to build the project (https://docs.platformio.org/en/latest/projectconf/section_env_general.html#projectconf-env-platform), etc.

And yeah, wrap it in a docker image, sure! :)

closed 2022-12-29 by wavexx