prusa3d/Prusa-Firmware #731: Temperature Calibration table

danielegger1, 2018-05-11:
I have used the temperature calibration from the newest firmware and I have seen that I will need less than 35° as well in the firmware.
When I start a "cold start" print (directly from SD card without heatup) the print start with Pinda temperatur e at 27° and my first layer is too high. When I warmup the printer for like 5 minutes the Pinda temperature is at 35° and when I start a print the first layer is perfect. I am printing in an ambient temperature (not measured from the printer) at 18° constanstly, so for normal PLA my Pinda does not heat up more than 37°.
So from my point of view the first Pinda offset should not be 35° it should start at 25° (and also adding 30 would be good).
Then I would be able to set the correct live z at 27° and after that I can modify the other values.

stahlfabrik, 2018-05-11:
I also think the range should be enlarged from 25C to 60C. 

Obviously the current workaround is to use the gcode „M860 S35“ in the start gcode to wait until pinda is warm.

But a quicker start without waiting for the pinda would be nice! So +1 for Daniel‘s request

Brainlazy, 2018-05-11:
Yes, a wider range of temperatures in the table wold be nice.
If you do use M860 then consider also heating the nozzle in two stages to avoid oozing while waiting for the pinda to reach 35c.

stahlfabrik, 2018-05-11:
I park the pinda so low for warming that the ooze forms a nice blob that sits on the heatbed. It is easy to remove during mesh bed leveling. So that is also a way to address this

BManx2000, 2018-05-12:
I noticed this issue myself and modified the calibration range to be 28 to 58 degrees:
https://github.com/prusa3d/Prusa-Firmware/pull/735

This is my first time editing the firmware, let me know what you think!

stahlfabrik, 2018-05-12:
I don’t know if it is good to lower the resolution as there are people where the calibration values jump quite strong from one temp step to the next. 
I personally would prefer extra steps than bigger steps:-)

But I have not fully thought this through

BManx2000, 2018-05-12:
You're right, I'm just not sure how to add new values to the EEPROM so I left it with the current 5.

stahlfabrik, 2018-05-12:
Hi!

It starts with configuration.h

#define EEPROM_PROBE_TEMP_SHIFT (EEPROM_PRINT_FLAG - 2*5) //5 x int for storing pinda probe temp shift relative to 50 C; unit: motor steps

stahlfabrik, 2018-05-12:
But a factory reset is needed to make the new layout active then I guess. 

So maybe one could save the extra steps to the end of eeprom list. So programming would be a little less straightforward. 

I do not know which is better! What do you think. Take your time to understand configuration.h a little first. :-)

closed 2023-09-04 by github-actions[bot]