prusa3d/Prusa-Firmware #732: 3.2.0 Wizard Live-Z "Try Again" bug

phooton, 2018-05-11:
Run Wizard on 3.2.0 final and when I repeat Live-Z test the extruder always seem to start from same height, but the shown Z offset stays at the value from previous run. Making the Z offset calibration unusable unless I hit it the very first time correctly?

stahlfabrik, 2018-05-11:
I guess yes. 

Do it only once. Then continue adjusting live z with a normal print. The Prusa Logo is fine for that for example

alhazar313, 2018-05-11:
Encountered the same issue

Panayiotis-git, 2018-05-11:
Same here, with MK2.5 v3.2.0. First run was -650, second -1320, third run was -2050.

Zaf9670, 2018-05-12:
Can you all provide which printer you are using as well? Just wondering if this is isolated to the Mk2/Mk2.5 or if it also affects the Mk3.

alhazar313, 2018-05-12:
Mine is a MK3 and has the issue

layneok, 2018-05-13:
Mine is also a MK3.

josepvm, 2018-05-13:
Same issue for me in a Mk3, after upgrading to 3.2.0 final.

PavelSindler, 2018-05-15:
Thank you for reporting this bug. PR https://github.com/prusa3d/Prusa-Firmware/pull/747 should fix this.

ram64, 2018-05-17:
I have the same issue with a MK2.5 upgrade. When using the Wizard to calibrate the printer I've been asked if I want to retry the Live Z adjust and I said "Yes". The operation started from the beginning with the nozzle height unadjusted, but the value remained from the first try. I had to double the value effectively to get the results. On the first print the nozzle dragged across the bed scraping it. I've had to reset XYZ calibration and start over.

Panayiotis-git, 2018-05-21:
I tested with the 3.2.1-576 release and I confirm that the issue has been fixed.
Please close the issue.

PavelSindler, 2018-05-22:
It was fixed in 3.2.1: https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.2.1

closed 2018-05-22 by PavelSindler