prusa3d/Prusa-Firmware #733: Pinda adjustment required

tbergman8, 2018-05-11:
After installing 3.2.0 final (same thing with RC2), I began getting mesh bed level failures. I never had these with 3.1.x. 
These were resolved by lowering the pinda a bit. It seems that the pinda sensitivity has changed with the new release.

stahlfabrik, 2018-05-11:
Sorry, I cannot confirm. Which should not mean you are wrong.

IgnacioHR, 2018-05-12:
When building the printer from the kit. The instructions tell you how to calibrate the distance between the extruder end and the pinda end. They use a flange to set that distance. What could we use as a reference to set the new distance between the pinda and the bed after installing 3.2.0?

PavelSindler, 2018-05-14:
Please re-run Z calibration in this case (menu "Settings -> Calibrate Z"). Results from mesh bed leveling are compared to last XYZ or Z-calibration results (Z-calibration is part of XYZ calibration and can be run separately) to detect possible failures.

PavelSindler, 2018-05-14:
However lowering PINDA probe is absolutely right.

AlGreeneye, 2018-05-16:
I also had this issue where it took me 3 goes at calibration for it to pass and then when I got to setting the live-Z I gave up at -1.4 where it was still too high and went to bed.  When I get home from work I will adjust the pinda lower and give it another go.

IgnacioHR, 2018-05-16:
@anyonewhohasloweredpinda can you provide feedback about the new distance between pinda and bed? Are the instructions in the kit still OK using a flange to measure that distance?

tbergman8, 2018-05-18:
My Pinda is currently approximately .8 mm above the nozzle and is working fine. After this change, I had to change live Z from about -600 to around -830

tbergman8, 2018-05-18:
Closed by mistake

closed 2023-09-04 by github-actions[bot]