prusa3d/Prusa-Firmware #736: Garbage output during self-test & xyz calibration on 3.2.0

zmerlynn, 2018-05-12:
I was running another self-test, and right after waiting for bed cooling (and possibly one other step), it went into this garbage display like the LCD was just displaying random memory. Freaked me out for a second, then the self-test continues. 

Then something similar happened during XYZ calibration.

I didn't see either of these, so I'm assuming it's something to do with (MK3) 3.2.0

Screenshots attached.

[image]

[image]

Zaf9670, 2018-05-12:
Did you compile your firmware using Arduino IDE or just apply the release file?

tuxmobil, 2018-05-12:
I've seen similar garbage before my MKS 2.5 upgrade and, as no other malfunction appeared, assumed that it is an interference problem with the display wires. Haven't seen this after rewiring during MKS2.5 upgrade.

lolsborn, 2018-07-01:
I had this happen during initial calibration once.  Restarting it seemed to fix the issue and I haven't seen i since.

closed 2023-08-24 by 3d-gussner