prusa3d/Prusa-Firmware #738: Full RX Buffer

vestom, 2018-05-13:
Just upgraded firmware on my MK3 to 3.2.0 and started having issues when printing from OctoPi over the USB cable.
The terminal transcript shows many messages like this:
Recv: Full RX Buffer
Recv: Error:checksum mismatch, Last Line: 849

It looks like the new firmware cannot keep up to process the serial buffer. Printing from the SD card runs fine.

Other users are having the same issue: [forum](https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/firmware-3-2-0-for-original-prusa-i3-mk3-t17799-s20.html#p84224)

Brainlazy, 2018-05-13:
Is it only on the ECOR_TOWER or have other files exhibited the same problem?

vestom, 2018-05-13:
I have exhibited it on other files.

rommer, 2018-05-13:
What baud rate are you connecting at?

Zawadaki, 2018-05-14:
I had put a waiting cycle to warm up pinda probe anh started to receive this kind of errors while waiting

MarekC3D, 2018-05-14:
We tested it with no issue. Send more information (Octoprint version, Gcode, Printer comm. setup...), please.

Zawadaki, 2018-05-14:
I added M860 S40 to o my start g-code right before mesh bed leveling

Zawadaki, 2018-05-14:
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N137 M105*34
Recv: P:39.9/40
Recv: P:39.9/40
Recv: P:39.9/40
Recv: P:39.9/40
Recv: P:39.9/40
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N138 M105*45
Recv: P:39.8/40
Recv: P:39.8/40
Recv: P:39.9/40
Recv: P:39.9/40
Recv: P:39.9/40
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N139 M105*44
Recv: P:39.9/40
Recv: P:40.0/40
Recv: ok
Send: N140 M105*34
Recv: Full RX Buffer
Recv: ok T:34.8 /0.0 B:85.1 /85.0 T0:34.8 /0.0 @:0 B@:38 P:40.0 A:38.6
Send: N141 G80*21
Recv: ok T:34.8 /0.0 B:85.1 /85.0 T0:34.8 /0.0 @:0 B@:38 P:40.0 A:38.6
Send: N142 G1 Y-3.0 F1000.0*47
Recv: Error:Line Number is not Last Line Number+1, Last Line: 15
Recv: Resend: 16
Printer requested line 16 but no sufficient history is available, can't resend
Changing monitoring state from "Printing" to "Error: Printer requested line 16 but no sufficient history is available, can't resend"
Changing monitoring state from "Error: Printer requested line 16 but no sufficient history is available, can't resend" to "Offline (Error: Printer requested line 16 but no sufficient history is available, can't resend)"
Connection closed, closing down monitor
Closing down send loop

PavelSindler, 2018-05-15:
Thank you for reporting this issue. M860 gcode should send "busy: processing" mesages during waiting. We will fix that. If you use Octoprint please add M860 to long running commands (Serial Connection -> Firmware and protocol -> Advanced) as workaround for now.

PavelSindler, 2018-05-15:
https://github.com/prusa3d/Prusa-Firmware/pull/744 fixes this issue.

stahlfabrik, 2018-05-15:
Thank you @PavelSindler. :-)

Zawadaki, 2018-05-15:
I managed to fix this problem temporarily by my own by changing setting in Octoprint
Serial port settings -> Firmware & protocole -> Simulate ok for resend requests -> Always

vestom, 2018-05-15:
I just tested with a simple test cube with no issue. But when re-testing with the ECOR_TOWER file the problem re-appears. However the file does not contain any M860 codes. 

When looking through the ECOR_TOWER file it looks like the culprit is the "TMC_SET_STEP_Exx" commands. 

18:08:05,866 - Send: TMC_SET_STEP_E26
18:08:05,869 - Recv: ok
18:08:05,872 - Send: N903 G1 Z2.800 F10200.000*55
18:08:05,875 - Recv: ok
18:08:05,881 - Send: N904 G1 E0.80000 F2100.00000*29
18:08:05,882 - Recv: ok
18:08:05,885 - Recv: ok
18:08:05,888 - Send: N905 M204 S800*98
18:08:05,891 - Send: N906 G1 F900*120
18:08:05,893 - Send: N907 G1 X137.275 Y117.275 E0.00609*97
18:08:05,894 - Send: N908 G1 X112.725 Y117.275 E0.83099*109
18:08:07,938 - Recv: echo:busy: processing
18:08:10,060 - Recv: echo:busy: processing
18:08:12,158 - Recv: echo:busy: processing
18:08:12,818 - Recv: ok
18:08:12,829 - Send: N909 M105*39
18:08:12,831 - Recv: Full RX Buffer
18:08:12,834 - Recv: tmc2130_goto_step 3 26 2 1000
18:08:12,837 - Recv: ok
18:08:12,842 - Recv: ok
18:08:12,844 - Send: N910 G1 X112.725 Y92.725 E0.83099*88
18:08:12,850 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 903

vestom, 2018-06-11:
Since upgrading to 3.2.1, I have printed lots of files from Octopi without issues. I have not retried ECOR_TOWER though. But if it is not a real issue for anyone, let us close this :-)

closed 2018-06-11 by vestom

jurriaan, 2018-09-18:
@PavelSindler I'm still having this issue on the latest firmware version on the MK3.

```
2018-09-18 20:37:38,874 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2018-09-18 20:37:38,931 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 108281, current line = 108286
| Last lines in terminal:
| Recv: ok
| Send: N108277 G1 X110.078 Y56.853*32
| Recv: ok
| Send: N108278 M204 S800*90
| Recv: ok
| Send: N108279 G1 F1500*127
| Recv: ok
| Send: N108280 G1 X110.057 Y56.806 E0.00136*106
| Recv: ok
| Send: N108281 G1 X110.060 Y55.922 E0.02302*108
| Recv: ok
| Send: N108282 G1 X110.129 Y55.826 E0.00308*110
| Recv: ok
| Send: N108283 G1 X110.208 Y55.761 E0.00268*100
| Send: N108284 G1 X110.502 Y55.596 E0.00879*110
| Recv: Full RX Buffer
| Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
| Send: N108285 G1 X110.773 Y55.525 E0.00730*97
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 108280
| Recv: Resend: 108281
```

also, as you can see in the log below, somehow the printer mixes in some of the commands send in the output.

```
2018-09-18 21:41:48,753 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 182939, current line = 182942
| Last lines in terminal:
| Recv: ok
| Recv: ok
| Send: N182935 G1 X175.553 Y57.634 E0.01273*98
| Send: N182936 G1 X175.563 Y57.876 E0.00631*105
| Recv: ok
| Send: N182937 G1 X175.563 Y93.567 E0.92997*101
| Recv: ok
| Send: N182938 M73 Q26 S524*27
| Recv: ok
| Send: N182939 G1 X175.586 Y103.426 E0.25688*81
| Recv: NORMAL MODE: Percent done: 26; print time remaining in mins: 51N182939  MODE: Percent done: 26; print time remaining in mins: 524
| Recv: ok
| Send: N182940 G1 X171.909 Y103.434 E0.09582*84
| Recv: echo:Unknown command: "24"(2)
| Recv: ok
| Recv: echo:Unknown command: "ok"(2)
| Recv: ok
| Send: N182941 G1 X171.488 Y99.703 E0.09783*103
| Recv: Error:No Line Number with checksum, Last Line: 182938
| Recv: Resend: 182939
2018-09-18 21:41:48,841 - octoprint.util.comm - INFO - Ignoring resend request for line 182939, that still originates from lines we sent before we got the first resend
request
```

alpha1125, 2018-09-25:
same error.

Send: N10677 G1 X149.710 Y99.034 E0.10047*90
Recv: ok
Send: N10678 G1 X149.114 Y99.128 E0.01609*87
Recv: ok
Send: N10679 G1 X148.984 Y99.166 E0.00359*93
Recv: ok
Send: N10680 G1 X146.908 Y101.242 E0.07828*110
Recv: ok
Send: N10681 M204 S1000*89
Recv: ok
Send: N10682 G1 X147.130 Y100.426 F10800.000*100
Recv: ok
Send: N10683 M204 S1250*92
Recv: ok
Recv: ok
Send: N10684 G1 F12000*118
Send: N10685 G1 X148.384 Y99.173 E0.04725*91
Recv: ok
Send: N10686 M204 S1000*94
Recv: ok
Send: N10687 G1 E-0.08000 F1800.00000*63
Recv: ok
Send: N10688 G1 F8640*67
Recv: ok
Send: N10689 G1 X147.130 Y100.426 E-0.35073*74
Recv: ok
Recv: ok
Send: N10690 G1 E-0.36927 F1800.00000*56
Send: N10691 G1 Z1.350 F10800.000*53
Recv: ok
Send: N10692 G1 X130.516 Y102.596*44
Recv: ok
Send: N10693 G1 Z1.050*117
Recv: ok
Send: N10694 G1 E0.80000 F1800.00000*16
Recv: ok
Send: N10695 G1 F1200*70
Recv: ok
Send: N10696 G1 X130.516 Y101.835 E0.01537*100
Recv: ok
Send: N10697 G1 E-0.80000 F2100*42
Send: N10698 G91*23
Recv: ok
Send: N10699 G1 Z0.600 F6000*29
Recv: ok
Send: N10700 G90*22
Recv: ok
Send: N10701 G1 X127.500 Y105.000 F10320*125
Recv: ok
Send: N10702 M400*35
Recv: ok
Send: N10703 M114*34
Recv: ok
Recv: ok
Recv: ok
Recv: ok
Recv: X:127.50 Y:105.00 Z:1.65 E:516.83 Count X: 125.01 Y:105.45 Z:1.75 E:544.85
Recv: ok
Send: N10704 G1 X130.516 Y101.835*37
Recv: ok
Send: N10705 G91*18
Send: N10706 G1 Z-0.600 F6000*55
Send: N10707 G1 E0.80000 F2100*15
Send: N10708 G90*30
Send: N10709 G1 F1200*66
Recv: ok
Send: N10710 G1 X130.280 Y101.504 E0.00822*100
Recv: ok
Send: N10711 G1 X129.999 Y101.167 E0.00885*98
Recv: ok
Send: N10712 G1 X129.999 Y102.453 E0.02596*109
Recv: ok
Send: N10713 G1 X129.481 Y102.453 E0.01045*96
Recv: ok
Send: N10714 G1 X129.481 Y100.458 E0.04028*96
Recv: ok
Send: N10715 G1 E-0.08000 F1800.00000*53
Recv: ok
Send: N10716 G1 F8640*69
Recv: ok
Send: N10717 G1 X129.481 Y102.453 E-0.39488*71
Recv: ok
Send: N10718 G1 X129.999 Y102.453 E-0.10248*77
Recv: ok
Send: N10719 G1 X129.999 Y101.510 E-0.18664*75
Recv: ok
Send: N10720 G1 E-0.03600 F1800.00000*62
Recv: ok
Recv: ok
Send: N10721 G1 Z1.450 F10800.000*56
Send: N10722 G1 X130.430 Y99.428*20
Recv: Full RX Buffer
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.

full gcode: https://gist.github.com/alpha1125/b0b5601a2bb261e6b848d2417b9f4bd9

jurriaan, 2018-09-25:
@alpha1125 Which firmware version, the latest?

I just had some prints printing fine using Octoprint by turning off filament runout detection, since that causes stutters (See #1187), could be due to the new 'Filament Engine'?

deejay2, 2019-03-31:
I have the same issue:

2019-03-30 18:25:14,107 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:25:44,186 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:26:14,239 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:26:44,286 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:27:14,365 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:27:43,213 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-03-30 18:27:44,407 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:28:14,486 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:28:41,265 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-03-30 18:28:41,308 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 152051, current line = 152057
| Last lines in terminal:
| Send: N152048 G1 X147.716 Y75.254 E0.17206*105
| Recv: ok
| Send: N152049 M204 S1000*108
| Recv: ok
| Recv: ok
| Send: N152050 M117 99% L=52/52*108
| Send: N152051 G1 E-0.08000 F1800.00000*5
| Recv: ok
| Recv: ok
| Send: N152052 G1 F8640*117
| Recv: ok
| Send: N152053 G1 X150.159 Y72.810 E-0.68400*64
| Recv: ok
| Send: N152054 G1 E-0.03600 F1800.00000*13
| Send: N152055 G1 Z10.800 F10800.000*50
| Recv: Full RX Buffer
| Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
| Send: N152056 M105*18
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 152050
| Recv: Resend: 152051
2019-03-30 18:28:44,524 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:29:14,596 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-03-30 18:29:16,149 - octoprint.util.comm - INFO - Finished in 13282.713 s.
2019-03-30 18:29:16,151 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2019-03-30 18:29:16,193 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2019-03-30 18:29:16,291 - octoprint.plugin - ERROR - Error while calling plugin tracking
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 230, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking/__init__.py", line 119, in on_event
    self._track_printjob_event(event, payload)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking/__init__.py", line 237, in _track_printjob_event
    sha.update(self._settings.get([b"unique_id"]))
TypeError: update() argument 1 must be string or buffer, not None

FormerLurker, 2019-07-18:
My printer is in the same room where I work, and I have been occasionally hearing the printer stop momentarily.  I usually chalk this up to my plugin, but this time I caught it in the act, and it stopped in a place that has nothing to do with my plugin (in the middle of a layer).  In fact, it just now stopped again as I'm typing this, which is really unusual.  I just checked and I'm not having any psu or overheating issues with my Pi3b+.

Here is some logging output from the first pause (a few seconds long):

```
| Send: N180752 G1 X146.222 Y122.616 E-0.27102*113
| Recv: ok
| Send: N180753 G1 X145.906 Y121.487 E-0.27075*118
| Recv: ok
| Send: N180754 G1 X145.822 Y121.061 E-0.10016*123
| Recv: Full RX Buffer
| Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
| Send: N180755 M105*25
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 180751
| Recv: Resend: 180752
```

Here's the second (more complete) log where a pause of approximately the same length occurred:
```
2019-07-18 16:35:05,149 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-07-18 16:35:05,180 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 186398, current line = 18402
| Last lines in terminal:
| Send: N186393 G1 X121.733 Y124.040 F10800.000*88
| Recv: ok
| Recv: ok
| Send: N186394 G1 F8640*117
| Send: N186395 G1 X121.754 Y124.070 E-0.02301*123
| Recv: ok
| Send: N186396 G1 X121.728 Y124.055 E-0.00687*125
| Recv: ok
| Send: N186397 G1 X121.580 Y123.717 E-0.08523*127
| Recv: ok
| Send: N186398 G1 X121.176 Y122.614 E-0.27111*118
| Recv: ok
| Send: N186399 G1 X120.860 Y121.486 E-0.27067*114
| Recv: ok
| Send: N186400 G1 X120.773 Y121.047 E-0.10312*116
| Recv: Full RX Buffer
| Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communicatin timeout if that happens regularly on specific commands or long moves.
| Send: N186401 M105*29
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 186397
| Recv: Resend: 186398
```
The full buffer and resend request is occasionally expected.  However, why the long (several second) pause?  This could be an OctoPrint issue related to the communication layer, but since it could also be an issue with the firmware caused by some lag responding to the resend request.  I wouldn't expect that to take seconds.  I'll report this to @foosel as well.

Is there any additional info that I could add?  I'm doing a lot of testing right now, and would be willing to gather more data.  I could also open a new issue if that's helpful.

Thank you for your time!

pczekalski, 2019-11-28:
Same here, Prusa MK3s, Prusa-Firmware 3.8.1 

I can barely print a thing now through RPI connected to it :(.

```
2019-11-28 15:00:43,821 - Send: N295 G1 F1500 X62.669 Y102.267 E69.41119*0
2019-11-28 15:00:43,825 - Send: N296 M205 X10 Y10*40
2019-11-28 15:00:45,339 - Recv: ok
2019-11-28 15:00:45,346 - Send: N297 M105*43
2019-11-28 15:00:45,348 - Recv: Full RX Buffer
2019-11-28 15:00:45,350 - Recv: ok
2019-11-28 15:00:45,353 - Send: N298 G0 F10800 X62.669 Y101.701*79
2019-11-28 15:00:45,358 - Recv: echo:Unknown command: "o@O`Cn OoC"(2)
2019-11-28 15:00:45,362 - Recv: ok
2019-11-28 15:00:45,366 - Send: N299 M205 X8 Y8*39
2019-11-28 15:00:45,372 - Recv: ok T:215.4 /215.0 B:60.2 /60.0 T0:215.4 /215.0 @:25 B@:7 P:34.1 A:37.1
2019-11-28 15:00:45,380 - Send: N300 G1 F1500 X92.796 Y131.828 E70.75741*6
2019-11-28 15:00:45,385 - Recv: Error:No Line Number with checksum, Last Line: 297
2019-11-28 15:00:45,388 - Send: N301 M205 X10 Y10*39
2019-11-28 15:00:45,390 - Recv: Resend: 298
2019-11-28 15:00:45,394 - Recv: ok
2019-11-28 15:00:45,396 - Send: N298 G0 F10800 X62.669 Y101.701*79
2019-11-28 15:00:45,401 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 297
2019-11-28 15:00:45,404 - Recv: Resend: 298
2019-11-28 15:00:45,406 - Recv: ok
2019-11-28 15:00:45,407 - Send: N299 M205 X8 Y8*39
2019-11-28 15:00:45,409 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 297
2019-11-28 15:00:45,410 - Recv: Resend: 298
2019-11-28 15:00:45,411 - Recv: ok
2019-11-28 15:00:45,412 - Send: N300 G1 F1500 X92.796 Y131.828 E70.75741*6
2019-11-28 15:00:45,414 - Recv: Error:No Line Number with checksum, Last Line: 297
2019-11-28 15:00:45,415 - Send: N301 M205 X10 Y10*39
2019-11-28 15:00:45,416 - Recv: Resend: 298
2019-11-28 15:00:45,417 - Recv: ok
2019-11-28 15:00:45,418 - Send: N298 G0 F10800 X62.669 Y101.701*79
2019-11-28 15:00:45,419 - Recv: ok
2019-11-28 15:00:45,420 - Send: N299 M205 X8 Y8*39
2019-11-28 15:00:45,420 - Recv: ok
2019-11-28 15:00:45,421 - Send: N300 G1 F1500 X92.796 Y131.828 E70.75741*6
2019-11-28 15:00:45,438 - Recv: Full RX Buffer
2019-11-28 15:00:45,698 - Recv: es�K����������������A"m=(u-o"(2)
2019-11-28 15:00:45,699 - Recv: ok
2019-11-28 15:00:45,700 - Send: N301 M205 X10 Y10*39
2019-11-28 15:00:45,708 - Recv: Error:No Line Number with checksum, Last Line: 299
2019-11-28 15:00:45,709 - Send: N302 G0 F10800 X93.362 Y131.828*74
2019-11-28 15:00:45,710 - Recv: Resend: 300
2019-11-28 15:00:45,712 - Recv: ok
2019-11-28 15:00:45,713 - Send: N300 G1 F1500 X92.796 Y131.828 E70.75741*6
2019-11-28 15:00:45,723 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 299
2019-11-28 15:00:45,724 - Recv: Resend: 300
2019-11-28 15:00:45,725 - Recv: ok
2019-11-28 15:00:45,726 - Send: N301 M205 X10 Y10*39
2019-11-28 15:00:46,899 - Recv: Error:Line Number is not Last������f84�ko�echo:Unknown command: "h-m	=kmr{---cm-- o?"(2)
2019-11-28 15:00:46,905 - Recv: ok
2019-11-28 15:00:46,907 - Send: N302 G0 F10800 X93.362 Y131.828*74
2019-11-28 15:00:47,161 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 299
2019-11-28 15:00:47,164 - Recv: Resend: 300
2019-11-28 15:00:47,168 - Recv: ok
2019-11-28 15:00:47,170 - Send: N300 G1 F1500 X92.796 Y131.828 E70.75741*6
2019-11-28 15:00:47,179 - Recv: Error:No Line Number with checksum, Last Line: 299
2019-11-28 15:00:47,182 - Recv: Resend: 300
2019-11-28 15:00:47,185 - Recv: ok
2019-11-28 15:00:47,187 - Send: N301 M205 X10 Y10*39
2019-11-28 15:00:47,188 - Send: N300 G1 F1500 X92.796 Y131.828 E70.75741*6
2019-11-28 15:00:47,196 - Recv: �vr[�line`Numbmv �����1�����������������X Last Line: 299
2019-11-28 15:00:47,198 - Recv: Resend: 300
2019-11-28 15:00:47,200 - Recv: ok
2019-11-28 15:00:47,201 - Send: N300 G1 F1500 X92.796 Y131.828 E70.75741*6
2019-11-28 15:00:47,217 - Recv: ok
2019-11-28 15:00:47,219 - Send: N301 M205 X10 Y10*39
2019-11-28 15:00:47,227 - Recv: Error:No Line Number with checksum, Last Line: 300
2019-11-28 15:00:47,229 - Send: N302 G0 F10800 X93.362 Y131.828*74
2019-11-28 15:00:47,231 - Recv: Resend: 301
2019-11-28 15:00:47,233 - Recv: ok
2019-11-28 15:00:47,235 - Send: N301 M205 X10 Y10*39
2019-11-28 15:00:47,241 - Recv: echo:Unknown command: ""(2)
2019-11-28 15:00:47,242 - Recv: ok
2019-11-28 15:00:47,244 - Send: N302 G0 F10800 X93.362 Y131.828*74
2019-11-28 15:00:47,254 - Recv: Error���)bb����9����300
2019-11-28 15:00:47,257 - Recv: Resend: 301
2019-11-28 15:00:47,258 - Recv: ok
2019-11-28 15:00:47,262 - Send: N301 M205 X10 Y10*39
2019-11-28 15:00:47,266 - Recv: ok
2019-11-28 15:00:47,268 - Send: N302 G0 F10800 X93.362 Y131.828*74
2019-11-28 15:00:47,276 - Recv: echK�����������������DF 1162"(2)
2019-11-28 15:00:47,278 - Recv: ok
2019-11-28 15:00:47,281 - Send: N303 M205 X8 Y8*37
2019-11-28 15:00:47,288 - Recv: Error:No Line Number with checksum, Last Line: 301
2019-11-28 15:00:47,291 - Send: N304 G1 F1500 X62.669 Y101.136 E72.12891*8
2019-11-28 15:00:47,294 - Recv: Resend: 302
2019-11-28 15:00:47,296 - Recv: ok
2019-11-28 15:00:47,298 - Send: N302 G0 F10800 X93.362 Y131.828*74
2019-11-28 15:00:47,302 - Recv: Error:No Line Number with checksum, Last Line: 301
2019-11-28 15:00:47,304 - Send: N303 M205 X8 Y8*37
2019-11-28 15:00:47,306 - Recv: Resend: 302
2019-11-28 15:00:47,308 - Recv: ok
2019-11-28 15:00:47,310 - Send: N302 G0 F10800 X93.362 Y131.828*74
2019-11-28 15:00:47,315 - Recv: Error:No Line Number with checksum, Last Line: 301
2019-11-28 15:00:47,318 - Send: N303 M205 X8 Y8*37
2019-11-28 15:00:47,319 - Recv: Resend: 302
2019-11-28 15:00:47,322 - Recv: ok
2019-11-28 15:00:47,324 - Send: N302 G0 F10800 X93.362 Y131.828*74
2019-11-28 15:00:47,327 - Recv: Error:No Line Num�+jRI�������f02
2019-11-28 15:00:47,328 - See octoprint.log for details
2019-11-28 15:00:47,335 - Changing monitoring state from 'Printing' to 'Offline: See octoprint.log for details'
2019-11-28 15:00:47,340 - Connection closed, closing down monitor
```

vestom, 2019-11-30:
Hi,
Looking at the log output, it does not look like a regular RX buffer overflow. That issue seems to have been solved since 3.2.1 and I have no problem printing from Octopi with firmware 3.8.1.

It looks like you are getting some serious interference on the USB line and it points more to a problem in your setup than a general issue. It could perhaps be due to a bad USB cable/connection? Have you tried another cable? Have you tried printing from a PC with Pronterface?

But please take it to the forums for further help, instead of posting here.