prusa3d/Prusa-Firmware #739: Turned of hotend and heatbed after several Y crashes, filament sensor not detected until printer turned off and on (reset was not helping)

pacaj2am, 2018-05-13:
Gcode was with LinearAdvance 

I did Extruder linearity correction 

I started print - was printing approx. 3 hours (unfortunately 9 Y crashes) , but when I came to the printer - both heatbed and HE was set to 0°C and printer was printing to the air (extruder was not jammed, when I preheated PLA i was able to get out filament). When I wanted to re-insert filament, filament sensor was not working, reset did not helped - I needed to turn of the printer, than it was able to detect filament sensor (reset did not help).

Zawadaki, 2018-05-14:
I had the same issue, filament sensor did not trigger when putting new filament

MarekC3D, 2018-05-14:
What is the status in Settings - Fil.sensor ?

Zawadaki, 2018-05-14:
Now is OFF but when this happed was ON

pacaj2am, 2018-05-14:
On my machine, there was N/A. When I tried to turn on, there was error message I don't remember.

Needed to power of printer and turn ON again.

japato, 2018-05-14:
I had the same issue as is on beginnig of this thread. Sems at mee, there is problem between 3.2.0 and Octoprint 3.1.8, see listing from Octoprint terminal, codes for cut-off temperature are sent to printer after crash message from printer. I had to dissable serial port on printer. Printing
 went from SD. Fillament sensor is always on.
[Crash.txt](https://github.com/prusa3d/Prusa-Firmware/files/2001766/Crash.txt)

zehkaiser, 2018-05-16:
I'm also having this issue. The first time the filament sensor was disabled and I walked into my office to see the HE and bed were turned off while the printer was still running. The second time I had enabled the filament sensor and it had a false detection of no filament. On resume both were off again. I have disconnected Octoprint (even though I've been printing from SD), disabled the filament sensor, and enabled stealth mode. I'll see if it happens again.

applekeith, 2018-05-16:
I just had my filament sensor replaced. Apparently, there was a bad batch. Mine was working fine, then went N/A and never came back. Their online chat resolved it quickly and efficiently.

pacaj2am, 2018-05-18:
I can confirm, that Heatbed and HotEnd were turned of after Y-crash detected. It happend yesterday aswell. Never happened before. I started to use Linear Advance with the new firmware.

jdn-za, 2018-05-20:
Having the same issues https://github.com/prusa3d/Prusa-Firmware/issues/754

PavelSindler, 2018-05-23:
Ok, It is clear to me now. This issue is caused by crash detection when using Octoprint (even if you are printing from SD and using Octoprint only for monitoring purposes).

Dolnor, 2019-01-17:
3.5.1. Had a safety timer pause on a 20h print pause itself at 18h mark via Octoprint, without losing printer communication. And it has decided it's gonna stop right then..

3d-gussner, 2020-07-13:
@pacaj2am This is an old issue and I think that has been fixed, Please confirm and close this issue.
[stale][fixed]

JohnnyDeer, 2020-08-12:
I'm closing this issue as stale, thanks for understanding.

closed 2020-08-12 by JohnnyDeer