prusa3d/Prusa-Firmware #740: MK3 Compilation with TMC2130_DEBUG Enabled

kickhad, 2018-05-13:
When I remove the comment from line in Configuration_prusa.h I get this compilation error.
#define TMC2130_DEBUG

Any other way to get M122 like debug info?

sketch/ultralcd.cpp: In function 'void lcd_main_menu()':
ultralcd.cpp:322: error: invalid conversion from 'void (*)(uint8_t) {aka void (*)(unsigned char)}' to 'menuFunc_t {aka void (*)()}' [-fpermissive]
         menu_action_ ## type ( args ); \
                                     ^
sketch/ultralcd.cpp:5621:2: note: in expansion of macro 'MENU_ITEM'
  MENU_ITEM(function, PSTR("recover print"), recover_print);
  ^
ultralcd.cpp:253: error: initializing argument 1 of 'void menu_action_function(menuFunc_t)' [-fpermissive]
 static void menu_action_function(menuFunc_t data);
             ^
exit status 1
invalid conversion from 'void (*)(uint8_t) {aka void (*)(unsigned char)}' to 'menuFunc_t {aka void (*)()}' [-fpermissive]

HFMan, 2018-05-17:
I can verify this problem exists for me as well.

devilhunter84, 2018-05-18:
Would like this also. 
https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/tmc2130-driver-infos-and-modifications-t15835.html#p79151

The feature has not been implemented from Marlin yet.

closed 2023-09-04 by github-actions[bot]