prusa3d/Prusa-Firmware #742: [MK3 3.2.0] - PID Calibration not suitable for HTPLA

workinghard, 2018-05-14:
While printing HTPLA, the nozzle is around 10C below the set temperature of 245C.
Reproducible issue. First layer is printed without a fan, nozzle keeps up at 245C. For the second layer as soon as the fan kicks in 100%, temperature drops by 10-15C. PID control reacts and trying to regulate but stays around 10C under the set temperature.
* Raising the temp leads to raised temp but still stays around 10C under the set temp
* Deactivating FAN leads to stable 245C
* Raising the FAN speed slowly (first layer 10%, second layer 20%.... to 100%) leads to stable 245C
* No issue on MK2S 3.1.0

I suggest to increase Kp and Ki values to make PID control more aggressive and compensate strong part cooling fan on MK3

kwaad2, 2018-05-14:
My PID is *PERFECT* and I don't wanna mess with it. (I get about 0.2c fluctuation)
But is it possible to do PID calibration with the fan on at 100%? That might help?

workinghard, 2018-05-14:
@kwaad2: Have you tried the described scenario? Does it work for you?

workinghard, 2018-05-14:
@kwaad2 Your screenshot does not look like perfect PID control :).
I have a workaround by raising the fan speed slowly over time. But that's not ideal solution causing the layer look slightly different. MK2S does not have this issue. It reacts a lot faster and stabilize at 245C already on the second layer. 
I can see PR is limiting a fan speed at 50% for their PET profile (240C). This is also one of the solutions to bypass this issue for HTPLA. But again, I see no reason why the PID control (or calibration) shouldn't be adjusted for MK3 to work with 100% Fan speed, even for higher temps.

PavelSindler, 2018-05-15:
Thank you for feedback.
Have you tried to run PID calibration? 
There are gcode M301 for setting Kp, Ki and Kd. You can do it manually using this gcode and then store it to eeprom using M500.

workinghard, 2018-05-15:
@PavelSindler: Yes, I ran the PID calibration without noticeable change. Is there any way to read the current values? Would be great to have a starting point for modifications. Will also try to run PID calibration with Fan on and report back the results.

kwaad2, 2018-05-15:
M503 will display your current values.
I am working on getting a PERFECT 230c (with part fan @100%).
I did a PID tune and it did good getting up to temp. (Slight hiccup at 236) but it's not holding steady enough for me. (Plus or minus 1.5c)
Restarted the PID tune, gonna take a while. I'll post my PID values when it's done. (assuming it's successful, although I have a custom fan)

Not sure if there is a PID tuned for every printer, or if it's just a "Default" value. But here is the PID that was on my printer, before I changed anything.
M301 P16.13 I1.16 D56.23

stahlfabrik, 2018-05-15:
Can you cancel? It is the old one IMHO

workinghard, 2018-05-15:
Done. Got the "newer" ones.

kwaad2, 2018-05-15:
https://preview.ibb.co/jd3fJy/PID.png

I tuned my PID a little bit.
I turned the part cooling fan, and then ran
M303 C50 S240
(this took a while)

I, like you tend to print "hot" and I also print with my cooling fan on.

My end result was
M301 P20.29 I2.26 D45.48
Recv:  Kp: 20.29
Recv:  Ki: 2.26
Recv:  Kd: 45.48

I do however have a different cooling fan nozzle than you, so my numbers would not be optimal for you.

workinghard, 2018-05-16:
Turning on the fan and do the PID calibration afterwards did the trick.
With regular PID calibration: Kp:15.73 Ki:1.14 Kd:54.00
After the Fan+PID calibration: Kp:17.73 Ki:1.34 Kd:58.47
[image]
Started printing, with the second layer temp wasn't able to keep up. Canceled the print. Did the enhanced PID calibration. Started the same print again. Temp is stable now with the second layer. 
Please consider to add this result to FAQ or/and adjust the default Kp,Ki,Kd values for MK3.

zafuquazar, 2018-05-16:
@workinghard what is “enhanced PID cal”?

workinghard, 2018-05-16:
Like described above. Turn the fan on to 100% (or maybe something 50-100% will work as-well). Then run the PID calibration. During the PID calibration you can't change the fan speed, so you have to do that prior calibration. This way the calibration logic determines more aggressive K* values. The only disadvantage is slightly less stable temp. In my case right now it's +/- 1C during printing.

zafuquazar, 2018-05-17:
Thanks, @workinghard!

workinghard, 2018-05-18:
I'm closing this issue because it can be handled with proper PID calibration or E3D sock. Thanks for working out the solutions. Further discussions can be handled in [this forum](https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/-howto-pid-calibration-your-way-t18479.html)

closed 2018-05-18 by workinghard