prusa3d/Prusa-Firmware #743: Compilation Error: 'distance_from_min' was not declared in this scope

stelgenhof, 2018-05-15:
The latest commit (553779a) fails to compile due to the following error:

`mesh_bed_calibration.cpp:2635:4: error: 'distance_from_min' was not declared in this scope`.

PavelSindler, 2018-05-15:
I see, it is MK2 variant. PR https://github.com/prusa3d/Prusa-Firmware/pull/745 fixes this issue. Please take in mind that 3.2.0 version was released just for MK2.5 and MK3. We would like to make new fw compatible with MK2, but it is not ready yet. If you want to play with MK2 fw, I would recommend to use 3.1.0 version from MK2 branch.

stelgenhof, 2018-05-15:
Thanks @PavelSindler ! Yes, I forgot to mention that I used the MK2 variant as I was actually checking if it was possible to compile an MK2 binary using the MK3 branch :) 

From the MK3 branch code base I thought I could also compile a MK2 binary. This is also why I submitted issue #730. But understand now it's not ready yet.

stelgenhof, 2018-05-19:
I can confirm the compilation issue has been fixed in the latest commit (92cb1dc3db2e22886b86da5c398a3bc83e9edfb9)

closed 2018-06-06 by PavelSindler