prusa3d/Prusa-Firmware #749: 3.2.0 Final Firmware BUG

MROJOD, 2018-05-16:
i just updated the 3.2.0.   10 may  firmware, when i slice any gcode on any print setting that includes linear advance it purges normally, prints brim normally, but as soon as it starts printing the piece it self it starts clicking when retracting until it finally pushes the filament all the way back, if i slice again deleting the m900 k line from the gcode and i select any print setting without linear advance it prints normally, i run latest version of drivers and only use prusa slicer( latest version) i was asked by the support chat to try many things like turning off filament sensor or try sample prusa gcodes wich prints normally but the problem always comes when i use linear advance settings, i was offerd to send the printer back but since i use it for work this is kind of a downside for me plues i really think its only a bug so why send it allt he way back right ? any help on this ? many thanks in advance

HFMan, 2018-05-17:
Very odd.  I don't have any issue with LA enabled.  Can you paste your exact start GCODE script?  And what slicer are you using?

HFMan, 2018-05-18:
Why are you editing the gcode direct?  Put the K value in the start script.

TheCodeSharman, 2018-05-20:
Sounds like the problem is in the per filament gcode rather than the start gcode. I'm guessing there is a "M900 K3330" command in the specific code for the filament. 

This defaults to something  like `M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{else}30{endif}; Filament gcode`

The code inserted depends on which filament is selected. @MROJOD check your filament settings?

MROJOD, 2018-05-22:
@Panayiotis-git  thanks for taking the time, as i said i did not edited o make any changes to the firmware or the slicer, as i have other 2 mk2s printers with 3.1.4.0 firmware both using the same slicer same settings and the only printer with 3.2.0 firmware was presenting the issue, the only gcode that came out different from the sd card was the one in the 3.2.0 mk2.5 printer, i am still using same slicer same settings same gcode as before and all printer are working fine, thanks for your input

closed 2018-05-22 by MROJOD