prusa3d/Prusa-Firmware #751: 3.2.1 Filament sensor error during self-test

danecreekphotography, 2018-05-19:
I've updated my MK3 to 3.2.1 from 3.1.1 and now the self-test is failing with a filament sensor error and wiring error. This appears to be the same error [reported in this post by william.t14](https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/firmware-3-2-1-for-original-prusa-i3-mk3-t18474.html#p86105).

The filament sensor worked fine on the previous version of the firmware.

danecreekphotography, 2018-05-19:
The fix for this, at least as a workaround, was simple: I turned the printer off and then on again and self-test passed.

I didn't ever power cycle the printer after flashing the new firmware and starting the calibration wizard.

danecreekphotography, 2018-05-21:
Another report in the forums: https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/firmware-3-2-1-for-original-prusa-i3-mk3-t18474-s20.html#p86428

stahlfabrik, 2018-05-21:
So the issue is resolved after power up? What is the issue then? Honest question:-)

danecreekphotography, 2018-05-21:
At least three people have reported hitting the problem in the forums, and then waste their time debugging trying to figure out why it doesn't work. Presumably an error after upgrading the firmware isn't designed behaviour...

danecreekphotography, 2018-05-21:
It can certainly be a decision to not fix this since there's a workaround that can be documented, but saying "what's the issue then" belittles the people who paid $800 for a printer because they expected it to be generally better quality than what else is out there, hit this, and then struggle.

stahlfabrik, 2018-05-21:
Understand. Maybe it is unfixable. But maybe the communication routine could be improved. 

I always do a factory reset after an update so I always start clean. The MK3 with its ruff release in November somehow taught me that lesson

gadams999, 2018-06-01:
I have the same issue on a new build, albeit I did have the cable backwards. Heading to the menu shows N/A for the sensor. Power cycling hasn't helped. I've reached out to Prusa for guidance (and cost on a new sensor if I blew it up)....

closed 2023-09-04 by github-actions[bot]