prusa3d/Prusa-Firmware #752: Add the capability to change the default preheat settings and save the changes

FmrAnimalTech, 2018-05-20:
Add the capability within the firmware to change the default preheat settings to the temperatures you want instead of the default provided. Allow the changes to be saved in the firmware so you don't have to keep changing them every time you power printer on.

Zaf9670, 2018-05-20:
You can easily change the filament temperatures if you compile the software yourself. And typically the preheat settings are close enough for the majority of filaments in those families when trying to load and unload filament.

On top of that, your slicer will let temperatures as part of the gcode. Do you have a very specific use case for this?

FmrAnimalTech, 2018-05-20:
Yes I do have a specific use case for this and the preset temps are to hot for my application. My printers are in full time production; I use different temp settings that the Prusa supplied settings. The open source firmware in my other printers allow me to make the changes I am suggesting in the enhancement request.  The user should not have to recompile the firmware just to change and save the default preheat temp settings. This should be an available option just as it is in other open source firmware implementations. You can take a look at how this was achieved and implemented in this firmware; https://github.com/andrivet/ADVi3pp-Marlin/releases

Zaf9670, 2018-05-20:
I'm not saying it cannot be added. But it's also more memory required for these already taxed 8-bit cores I would assume. You also have alternative options assuming you're running in a farm type capacity and use monitoring software.

Typically things like Octoprint and Reptier offer ways to set designated preheat settings. And like I said before, normally your slicer will handle temp changes as well in the gcode. Just trying to offer alternatives. I rarely use the preheat settings on the printer itself other than filament swapping.

FmrAnimalTech, 2018-05-20:
Zaf9670, thank you for your feedback. at this point I don't use Octoprint or Reptier I am doing everything via SD Card which is the simplest way for my process. I have each printer plugged into their own pure sine wave UPS and have limited as many failure points as possible to keep my production running. In the future this may all change based on volume requirements, space and so on. Being able to walk in front of my printers, turn on preheat for the filament I am using, clean the nozzle once it reaches the desired temp and when I hit print the printers start immediately is a process that works for my situation very well. This has reduced my instances of clogs and failed prints to almost zero. Having this feature enabled just helps my process remain efficient and the ability to adjust through the menu system as filament requirements change is also a very desirable scenario. My goal is to spend my energy designing and making prototypes then print thousands of those parts. My printer should be able to help me in this process and not require me to recompile the firmware every time I need to make a slight change. I am sure there are others who use this feature in other printers as well and find it very useful. It can only help this printer deliver the best features to the end user which I am sure is the goal of the manufacturer.

Zaf9670, 2018-05-21:
I don't disagree with the ease of use portion you're describing. I honestly don't know much about the architecture of the hardware and firmware restraints at the moment but I did see the recent commits regarding flash memory being recovered so that's why I voiced the limits earlier.

One thing you could consider is making the startup script take care of the nozzle cleaning for you potentially. I believe some people have installed small brass wipe brushes onto beds in the past. But as far as your preheat goes I don't know what you can do for an immediate resolution other than compiling yourself.

KD7LRJ, 2018-05-21:
Would it work to create a gcode file and run it from the card instead of using the preheat option? Seems like it could move, clean, and heat according to your needs.

Zaf9670, 2018-05-21:
I guess I didn't think of that myself. I'm so used to the ending scripts containing the motor and heater off commands I forgot they're not required.

FmrAnimalTech, 2018-05-21:
I do agree that some sort of gcode commands in the starting script to preheat the extruder and bed prior to the "G28 ; home all axes" command would work assuming that the MK3 would execute the command with the extruder and the bed cold at the beginning. I have not tried this as I have not seen an example of the commands in question. Is it possible to execute this and tell the printer to wait until the desired temps are reached for both extruder and the bed before continuing?

For PLA I use Extruder 205 and the Heated Bed 65  
For PETG I use Extruder 250 and the Heated Bed 75

closed 2023-09-04 by github-actions[bot]