prusa3d/Prusa-Firmware #754: 3.2.1 - Temps lost after crash detection

gdombiak, 2018-05-20:
### Setup
Firmware: 3.2.1 (did factory reset and redid calibration including e-correction)
Printer: MK3
Settings: Crash detection [on], Filament sensor [off]
Octopi: Connected but just for recording timelapse and monitor temperatures

### Steps
While I was printing [MG_SleepingDog.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/2020053/MG_SleepingDog.gcode.zip), when I was at 92% there was:
1. a crash detected on Y axis (my first crash ever and no idea why it happened)
1. Printer printed "crash detected" on the LCD
1. Nozzle was moved to the left side of the printer (maybe it went to 0,0 but not sure)
1. It resumed printing immediately (on its own)
1. I noticed that the display was showing 0 for target hotend temp and 0 for target bed temp
1. I quickly set the temps to what I was using (220 for hotend and 60 for bed)
1. Printer resumed printing

jdn-za, 2018-05-20:
Been having the same issue on Y crash (no apparent actual crash) on FW 3.2.0

floyd871, 2018-05-21:
I can confirm this issue. Running on FW 3.2.0. After an hour of printing i have noticed that that the printer is still printing but without filament. The temperature setting is "0" (hotend/bed). I was not in the room when a possible "crash" was detected. After Downgrade to FW 3.1.3 i saw sometimes a "Crash Detection" but the temperatures stays as they are before.
So may be two issues here:
Too sensible crash detection and disable temperatures....

gdombiak, 2018-05-23:
This is not fully reproducible. I had 2 more Y crashes and those 2 times the temp was preserved. Not sure how to reproduce this issue.

PavelSindler, 2018-05-24:
During crash detection, printer behaves the same as if print is stopped. Octoprint is sending M27 periodicaly to get printer status (which byte from file is currently printed etc.). If M27 and crash detection happens at the same time, printer sends "Not SD printing" to Octoprint and Octoprint sets printer temperatures to zero. I made fix which is just sending different status to Octoprint in this case. It is part of PR https://github.com/prusa3d/Prusa-Firmware/pull/760 .

gdombiak, 2018-05-24:
Pavel,

Thanks for the detailed explanation. Do you want me to close this ticket then since it is getting resolved in that PR?

closed 2018-06-06 by PavelSindler