prusa3d/Prusa-Firmware #755: Z level is lost on restart

gdombiak, 2018-05-20:
### Setup
Firmware: 3.2.1 (did factory reset and redid calibration including e-correction)
Printer: MK3

### Steps
1. Move Z axis to some place (in my example it was at 25)
1. Turn off printer using PSU switch
1. Wait 30 seconds or more
1. Turn on printer
1. Z level is at 0.000. Expected to be at 25

To correct this I do Z calibration, or use Octopi to move the Z axis up to 210 (reaches top and motor keeps spinning since it things it needs to go higher), or I use the knob to move Z to 210 same as I do with Octopi

Zaf9670, 2018-05-20:
Unless there's a feature I'm not aware of this is normal behavior if you're not running a print and why G28(I think that's homing) is always part of the starting gcode scriot.

Also, the power switch is not the proper way to test the power fail on the Mk3 if that was your intent.

gdombiak, 2018-05-20:
I’m not testing power fail. I’m just turning off my printer once I’m done printing. :)

The problem is that it thinks that z = 0 is in the wrong place. Only way to reset it is how I described in the issue.

stahlfabrik, 2018-05-20:
All you have to do is home the machine. That is what homing is for. 

You do not want to save every change of z to eeprom, which has limited write cycles. 

No issue here?!

gdombiak, 2018-05-20:
So the process is to **manually move the nozzle to 0,0** before turning it off? 

I tried finding the procedure to turn-off the printer on the handbook and other places with no luck. Or I need to improve my search skills or the manual needs to include this info. :) Your EEPROM explanation makes sense about saving z value to each time. Will close this ticket as soon as I confirm correct shutdown procedure.

Panayiotis-git, 2018-05-20:
In fact, this is the case for all 3 axis. Not only Z. When you power on the printer, it does not know where the hot end is. Even if during shutdown the printer stored the known position, and then while it was closed the user could move the extruder to another position, the stored position would be invalid.

The solution, as been told previously, is the homing function. Whenever the printer starts a print, there is a command in the g-code file, that instructs a homing. After this, the printer knows the exact position of the extruder.

So, you do not have to take any shutdown procedures.

stahlfabrik, 2018-05-20:
I also do not understand why you would manually move  z down - regularly. 

As described above, the printer orients itself when it needs to - right before a print

Zaf9670, 2018-05-20:
The other reason for this is due to printer startup not having the stepper motors engaged. Since you could have moved the bed after shutdown.

While it's less likely you moved your Z-axis that's another reason why I would assume it doesn't save position during shutdown or power outages. Even the power panic re-homes before resuming.

HFMan, 2018-05-20:
This should be closed... this is NOT a bug- it is working as designed.  Power panic indeed DOES save the current z-position (as it can't move the nozzle back down to the plate during the rehome).  But turning off the power does not save any axis position.  And before asking "why doesn't it save this every second or so", you would burn up the eeprom writing this information over and over.

gdombiak, 2018-05-20:
Thanks guys for the educational information. I understand now how things work. I will close this ticket since there is nothing wrong with the firmware. I think that adding this information to the handbook will help other newbies understand that this is not a problem.

closed 2018-05-20 by gdombiak