prusa3d/Prusa-Firmware #757: MK3 3.2.1 Nozzel

trogdor88, 2018-05-20:


Zaf9670, 2018-05-20:
I know some people have said they needed to lower their PINDA probe on a recent update. And it could also be the temperature calibration on the PINDA causing the issue. May want to disable that and try again.

trogdor88, 2018-05-20:
I did re-verify the PINDA height as one cable tie.  I can lower it, but the interesting thing is after the crash and subseqent calibration failed if I long press the control for a live z adjust it shows a positive value.  I will disable temp calibration and try again before I move the PINDA.  Thanks.

Zaf9670, 2018-05-20:
I decided to put my PINDA probe at pinching the zip tie close due to my head crash issue. It's better to have the probe closer than further as others have discussed on here.

I was also just suggesting the temperature offsets could be right on their tipping points but I think the general consensus has been that the manual calibration should be performed for best performance.

What were you heated to when it crashed? PLA or something hotter?

kroll999, 2018-05-20:
@trogdor88 same problem here. 
Confirm test was successfully on 3.2.0

trogdor88, 2018-05-20:
Just to clarify I was not printing at the time.  I was only running the temperature calibration under the calibration sub menu.  I did have PLA loaded but as the nozzle was cold [25,26]C and clean.

vaxxi, 2018-05-21:
Can confirm the problem on 3.2.1. Started temperature calibration (steel sheet removed), nozzle lowered until it was touching the heatbed, when moved for the next step it was dragging on the heated bed. Reset printer, Z coordinate was indicated as 0.15 even if the nozzle was touching the heatbed (with the steel sheet removed).

klonk, 2018-05-23:
Same for me. 3.2.1 on mk3 Nozzle  just dig to the heated plate

PavelSindler, 2018-05-24:
Dont you know if homing at the beginning of temp calibration happened with PINDa probe right above calibration point?

trogdor88, 2018-05-24:
It did.  In my case it is on the start of the second test. I have yet to move the PINDA as @Zaf9670 suggested so it is 100% repeatable for me let me know if you need a video or anything.

PavelSindler, 2018-05-30:
I made PR with temp calibration improvements: https://github.com/prusa3d/Prusa-Firmware/pull/788

trogdor88, 2018-06-07:
Thank you!  Once I finish printing I'll build and give it a shot.

chankinfun, 2018-06-09:
I also have this problem, but no problem on 3.2.0

sidoh, 2018-06-18:
This also happened to me (picture attached).  It also didn't complete -- was stuck on 6/6 for about two hours before I gave up and reset.  I can try again on 3.2.0, but I'm a little fearful for my heated bed!

[image]

trogdor88, 2018-06-29:
All working with MK3 running 3.3.0-RC1.

closed 2023-08-24 by 3d-gussner