prusa3d/Prusa-Firmware #758: 3.2.1 - Enclosed ABS Print, Heatbed & Hotend set 0 after 3h while XYZ motion continues

TheZeroBeast, 2018-05-21:
##FW: 3.2.1
Enclosed ABS Print, Heatbed & Hotend set 0 after 3h while XYZ motion continues with serial communication advising "cold extrusion protection activated"

Why did the heaters shutdown is the question.

Appears to have started happening since SafetyTimer implementation in FW.

Gcode attached bellow with screenshot of failure position and temp graph of failure.

[RAPTORMixed.zip](https://github.com/prusa3d/Prusa-Firmware/files/2021248/RAPTORMixed.zip)

[image]
[image]

I will attempt to gather log files for time of print and attach.

TheZeroBeast, 2018-05-21:
I think there is a trend with other reported issues perhaps.
#734
#720
#739

TheZeroBeast, 2018-05-21:
[image]

Zaf9670, 2018-05-21:
@TheZeroBeast Do you know what layer it froze up?

TheZeroBeast, 2018-05-21:
Hi @Zaf9670, heaters turned off just past the 3h mark of the print at about 4mm in height which led the extruder to stop when hotend dropped bellow mintemp of 180-190C which ever the MK3 FW is using at present.

TheZeroBeast, 2018-05-21:
As others have suggested this issue is happening when a crash is detected I ran a test PLA print and induced some crashes although it worked as expected without shutting off heaters.
This leads me to think it is time dependant.

Zaf9670, 2018-05-22:
So are you suggesting this issue is do to a false positive on the power panic? Or that you may have had a brownout/blackout that caused this to happen? Just following this in case it is something that could happen to a long running print of my own.

Side note: You can try resuming the print from where it left off next time. Roughly find where your print ended and tell the slicer to begin at that height and continue the print. The poor man's fix to a filament runout midprint.

TheZeroBeast, 2018-05-22:
Hi @Zaf9670, as per the forum post which is linked above there could have been an issue with the nAC_FAULT input and on @nicholas.g9 board the voltage drift suggests this. Voltage dropping to a point where half the hardware triggers results in heaters off and print continues. Error state indeed and not normal operation.

Once heaters are off there is at least 15m of printing plastic that increasingly doesn't adhear properly to previous layers so recovery of most prints would be a waste compaired to prints that have just stopped e.g. power loss on MK2.

TheZeroBeast, 2018-05-23:
I also just identified issues with the pause and resume functions M25 & M24.
Print pauses and kills heaters and disengages steppers.
Resuming print states on LCD ".....heating" although never sets target values. When set manually through menu and at temp again printer homes Z axis before promptly running head long into what was already printed. Luckily no damage from what I can see but not happy.

Same 3.2.1 FW on MK2.5

In my mind pause print would consist of pause, cool hotend, saving X,Y&Z values, move Y Bed forward, hotend to the side. Safety Timer disabled so print doesn't detatch with cooled bed.

Resume would consist of heating, homing of X,Y and resuming at saved X,Y&Z values.

jdn-za, 2018-05-23:
I have managed to do a 41 hour print with crash detection off so I expect the issue is there?

PavelSindler, 2018-05-24:
Yes it was caused by crash detection. Explanation is here: https://github.com/prusa3d/Prusa-Firmware/issues/754

TheZeroBeast, 2018-05-24:
Hi Pavel,
thank you for getting to the bottom of this issue. I look forward to testing it.
Rob

closed 2018-06-06 by PavelSindler