prusa3d/Prusa-Firmware #764: MK2.5 3.2.1 XYZ calibration fails

gregsaun, 2018-05-23:
I have upgrade from 3.1.3 to 3.2.1 and was unable to finish XYZ calibration. First two points where good but the third one (first on back) failed, unable to find it. I tried to lower the PINDA and did not solved the problem. Same results from wizard or from calibration menu.

I have rolled back to 3.1.3 and XYZ calibration is successful with following values :
- Measured Skew:  0.00°
- Y distance from min : 
  - Left : 6.07
  - Right : 6.27

Panayiotis-git, 2018-05-23:
I had the same problem with any MK2.5 firmware release. 
The solution for me was to move the Z frame about 5mm to the front.
Now, I have Left: 9.12mm / Right: 9.09mm / Skew: 0.00

I've also noticed that the 3rd point is a bit crowded. The noise pattern that the new calibration algorithm produces over the test points is a lot different at the 3rd point than the others. The other 3 point produces a very clean pattern.

gregsaun, 2018-05-23:
Thanks for the info Panayiotis. I read about this 5mm delta but could not find the issue anymore. Observing the calibration makes me feel that it is the same issue than you, all circles are done few mm in the back.

However I could not observe the noise pattern you are talking about, but maybe it is because I associate this to the fact it cannot get the third point.

gregsaun, 2018-05-23:
Here it is, similar issue since 3.1.4 :
https://github.com/prusa3d/Prusa-Firmware/issues/616

Grawp, 2018-05-26:
I seem to have had a same/similar problem. I have a completely different frame https://www.thingiverse.com/thing:1692666 (with MK2.5 parts and 3.2.1 firmware).

When I first upgraded form 'MK2-X' to 'MK2.5-X' I calibrated on the first try just fine on 3.2.0 firmware. Then later after some moving of the printer and upgrade to 3.2.1 I wanted to redo the XYZ calibration and it started to fail. A few times on point 1, a few times on point 4 and countless times on point 3.
So I tried lowering the PINDA probe, I even tried re-crimping its connector.

Then I remembered that when I first calibrated the XYZ, the PSU was not attached yet to the frame and it had been just lying on the table. So I removed the PSU from the right corner and the XYZ calibration started to work much more reliably. First time it failed again on point 1. but three more consequential calibration passed OK.

Could somebody try to dismount the PSU from the corner near point 3 and confirm that the calibration starts succeeding more often than not, especially not stopping on the 3rd point more often than on others?

triplepoint, 2018-05-27:
I ran my mk2 (converted to mk2s) with the PSU unbolted from the Y-axis frame for a long time, in order to be able to move the vertical frame forward on the threaded rods by 3 or 4 mm.  I wrote it up in more detail here:
 https://github.com/prusa3d/Prusa-Firmware/issues/616#issuecomment-385211047

Recently, I printed a 4mm thick spacer to go between the PSU and the vertical Z frame, which moves the PSU back enough to be able to bolt it back to the Y-frame, and still maintain >10mm values on the "Y distance from min" menu.

But yes, I definitely had trouble with the 3rd point, both with the recent 3.2.x firmware after my mk2.5 upgrade, and also before, with the 3.1.0 mk2s firmware.

Efroymson, 2018-06-13:
I had the same problem, was able to get past the "third point won't calibrate" problem by loosening the bolts holding the PSU.  I tried going back to the 3.2.0 firmware, since it looked like that had helped someone, but it did not work for me.  I tightened the PSU back, which I hope is the right thing to do

Panayiotis-git, 2018-06-13:
@Efroymson These bolts keep the printer's z-frame in a stable vertical position. But there is no room for adjustment. If you have your frame in a different from the "ideal" position then those screws bend the frame and the result is incorrect calibration.
With the MK2.5, the only way to succeed the calibration was to move the frame forward about 4...5mm (I guess bug in firmware). At this position, there is no way to keep the bolts.
You have to modify and reprint the PSU holder part.

Grawp, 2018-06-13:
@triplepoint and others: I haven't suggested removing the PSU so the frame could be adjusted. I've suggested it for removing source of interference for the PINDA probe (i.e. PSU is big feromagnetic metal block full of various electric currents).

Panayiotis-git, 2018-06-14:
@Grawp You have a very valid point. But, if the problem was the PSU, then the second point would have also a very high rate of failures.
As I mentioned above, there is some design issue around the 3rd point of the bed.

craiglink, 2018-12-26:
I’m having the same issue after upgrading my mk2/s to 2.5 and using the 3.51 firmware. It either fails on the 3 point or after trying all 4 points. When it moves on to all 4 points, can see that’s the location it thinks the 3rd point is at is incorrect.  Re-flashing with firmware 3.2.1 successfully and correctly calibrates.

closed 2023-09-04 by github-actions[bot]