prusa3d/Prusa-Firmware #765: Extruder Clicking with FW 3.2 on MK3

tyleryoung1230, 2018-05-23:
I assembled my MK3 last week. It printed perfectly for three days, and all of a sudden the extruder began clicking, skipping steps, and leaving a significant impact on my prints, up until the point now that it is basically unusable. I've reached out to Prusa support, and tried all of the adjustments with idler screws, and unclogging the nozzle. I've read other issues (695 and 696) that have a similar issue, but not identical. 

Thanks so much in advance. 

[image]

iAmZuit, 2018-05-24:
I've removed most of the extruder skipping on my MK2.5 by making sure the gears are lined up and tensioned properly, then adding a little bit of grease to the teeth of the bondtech gears. If they are out of round even a little they will bind and the grease seems to help that.

How I discovered that mine were binding was by heating the nozzle, disabling the steppers and pushing the filament through manually. It would go easily for a little bit and then get really hard. I'd push past that and it would get easier again. May be worth trying.

matthew-humphrey, 2018-05-25:
@iAmZuit when you say "gears are lined up" do you include the idler gear in that as well? If doesn't seem possible to like that one up on my MK3 due to the plastic cover part which forces it off to one side slightly.

daveespo, 2018-05-25:
Sounds related to #769

Skywatchr, 2018-05-27:
I can confirm this one. It might be a problem of the Bondtech gear becoming too hot. I probed 66°C after 7mm Z-height of the Marvin included on the SD-Card - and then it is when the skipping starts.

Stoobs, 2018-06-03:
I've got the same issue on my MK3, sometimes it can't even print a Benchy (<2 hours printing), other times it will just about complete, but with quite a few layer skips - I've adjusted the engagement screws to the point where just a quarter turn will lead to insufficient grip on the filament, so I know it's not an over-tightening issue.

Print speeds don't appear to make much difference, the extruder will still overheat and start skipping midway through a print.

I also own a MK2 which I've upgraded to a MK2.5, and while I do see an increase in extruder temperature, it's nowhere near as severe as the MK3, and I can happily leave it printing overnight without issue.

The extruder on the MK3 by comparison gets very nearly too hot to touch, so I'd estimate up around 80 degrees Celsius from checking against bed of the MK2.5 printing PETG...

Really frustrating because I know the MK3 can be so much better than this - I'm printing off the Bear extruder parts with the fixed filament path at the moment and will see if that solves the issue.

tyleryoung1230, 2018-06-03:
Please keep us posted on your progress!

Sent from Astro <https://www.helloastro.com> for iOS

Dioblieum, 2018-06-04:
I’m experiencing this also.  Started a couple weeks ago and makes any print over an hour or so fail.  Clicking starts about 1-2 hours into a print, eventually  extrusion stops completely, though clicking continues ;)  Tried varying speed, nozzle temp, idler tension screws, replacement of heatbreak.  Filament verified by using in a different machine (Formbot). Ambient temp is 29-33.  No enclosure.
I can seemingly reduce clicking by aiming a fan at the extruder, but it is only a temporary reprieve.  Any suggestions besides trying one of the better aligned extruder modifications others are working on?

Stoobs, 2018-06-04:
The only other option at the moment is I think to drop the current to the extruder down further - but surely we're just going to run into issues with that too. Has anyone compared the extruder code from before the 3.1.3 release with the latest to see if there's something odd going on with how the stepper motor is being driven?

JelteMX, 2018-06-06:
I'm not enitrely sure, but it seems that [this commit](https://github.com/prusa3d/Prusa-Firmware/commit/964d019dd18d269203b9124f6f07c21ce78428c6#diff-d1d79099d6fb1bd5281c9ea0deb5060c) is responsible for setting the max current for the Z stepper motor. Increased from 26 to 30. I'm not sure what the difference is, but this part has changed over the past releases. Since 3.2 I am getting under extrusion, where all my prints fail. Extruder stepper is very hot, way more than it was before the firmware upgrade

AbeFM, 2018-06-16:
Subscrbed.

I've had flexible filament shoot out the side, bend up my PTFE, and can't get flex working again until I replaced it.

Now I'm having trouble with Prusa PLA. Hard to say where it's from. Greased gears recently and everything was fine for a couple weeks.

plankton99, 2018-07-08:
i am having same issue with both MK3 and MK2 after a few firmware updates 3 months back.  i have troubleshooted with Prusa for 3 months, ordered a new fully assembled hot end as they said that was the problem, they sent me a new extruder motor, did firmware updates and software resets and still same problem of skipping and jamming.  This is a horrible experience.  Extruder gets so hot that common sense will tell you that it pre-melts the filament and that is what causes the problems.

bestgustavo30, 2018-07-08:
@plankton99 can you tell me what is the temperature of you heated bed when it's at 60° in the firmware??, mine don´t pass 49° or 51° I use the same thermometer.

Thanks :)

plankton99, 2018-07-09:
@bestgustavo30  when by bed is set to 90C the thermometer reads 80C, why do you ask?

matthew-humphrey, 2018-07-09:
The temperature reading is taken from a thermistor on the bottom of the heatbed. If you're measuring it on top, it's expected that these two temperatures will be significantly different.

plankton99, 2018-07-17:
Why does the extruded motor get so hot, compared to the other motors?  That’s the problem here...

matthew-humphrey, 2018-07-17:
Because it's operating without any gear reduction or other form of mechanical advantage.

plankton99, 2018-07-17:
The y-axis is not geared, it runs on a belt and doesn’t get hit and it has to move a lot more weight than a filament.

Regards,

Kamal A. Rajan

Skywatchr, 2018-07-17:
@plankton99 try to push some filament through the extruder by hand and then try to move the heatbed also by hand. You should be proven wrong :-)
Anyway you can try to tweak the current settings for the extruder motor in the firmware, have the idler 
door screws less tightened or use some funny cooling solutions from Thingiverse.

MrPigglesworth, 2018-08-14:
OK... I signed up for GitHub just to help everyone out with this issue...

For me the problem was using a Ruby nozzle... once I changed it out... no more problems!!!

My printer never really clogged with the Ruby nozzle... it would click and then stop extruding but there was never any obstruction...  but if I made the printer change out the filament( take the filament out... snip the end off and start printing again with the same filament) the printer would work for a while but soon it would start clicking again.

A couple of times after replacing the nozzle I thought the clicking was back but when I checked my temp I noticed it was way too low. Once I brought the temp back up... no more clicking.

And yes I checked my temp back when I had the Ruby nozzle installed.

It may not be because it was a Ruby nozzle but it definitely was a nozzle problem

Good luck... I hope this helps... I tore my hair out trying to get my printer working correctly again.

BlindSniper35, 2018-08-18:
Check your filament tension. I've had extruder skipping due to the PTFE tubing in my drybox causing too much resistance (I ordered some Capricorn to replace it). Pull some filament down off the spool and see if it continues to skip. if the skipping goes away that's your problem.

MrPigglesworth, 2018-08-18:
I changed out the Ruby nozzle and have had no problems since... again I may not be because it was a Ruby.

AbeFM, 2018-09-14:
> For me, the problem was fixed by replacing the extruder motor... for a
> while. I'm on my third one since May. It seems like whenever I have a
> long-running print jam partway through, the extruder toasts itself after a
> few hours of struggling against the jam.

It sounds dumb but I put a heat sink on the motor, it helps. Just zip tied
it on. For long prints on hot days I've tried a fan but I took it off again.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Prusa-Support, 2022-05-07:
This thread has been silent for years, moreover, some legacy FW and design upgrades have presumably addressed most of the aforementioned concerns by now. Therefore this issue will be closed for now.

__Design upgrade that improves the hotend cooling
- MK3S to MK3S+ design upgrade
https://www.prusa3d.com/product/original-prusa-i3-mk3-s-to-mk3s-upgrade-kit/ (complete)
https://www.printables.com/model/61392-i3-mk3-to-mk3s-upgrade-printable-parts (only print files for MK3)
https://www.printables.com/model/61346-i3-mk3s-to-mk3s-upgrade-printable-parts (only print files for MK3S)

__FW releases that improved the extruder stepper motor thermal conditions
- https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.2.0
- https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.10.0

Other possible complications that may be a cause for the extruder motor to click (skipped steps) are mostly covered in the following articles.
- about under-extrusion. https://help.prusa3d.com/en/article/under-extrusion_2007
- in case under-extruding starts on the first layer ttps://help.prusa3d.com/en/article/first-layer-calibration-i3_112364
- to prevent friction opposing the filament movement https://help.prusa3d.com/article/spool-placement_112035
- about heat-creep https://help.prusa3d.com/en/article/extrusion-stopped-mid-print-heat-creep_1948

This issue will be closed. Please comment below or open a new issue if you think the issue is not resolved.

There could be other causes for the issue not covered here, so please consider reaching out the Technical Support before posting further related comments.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

closed 2022-05-07 by Prusa-Support