prusa3d/Prusa-Firmware #772: PowerPanic does not work on 3.2.1

jonnieZG, 2018-05-28:
PowerPanic feature does not work on 3.2.1:

After resuming from a PowerPanic event, the bed and the nozzle heat up, head goes to last position and stays there, nothing happens. 

I would like to hear from someone who tested it successfully with release 3.2.1, if any.

Cheers,
Jonnie

PavelSindler, 2018-05-28:
Were you printing from SD card?

PavelSindler, 2018-05-30:
Can you provide gcode file?

tbedore, 2018-06-03:
Hi, I just tested this on a gcode I was running on my machine, printing from SD card everything worked fine, print resumed perfectly.

tbedore, 2018-06-03:
Though after the test was ran, the timer for how far into the print had reset, and there is no text showing where it would usually say printing this object

Prusa-Support, 2022-05-17:
Hello! If you are still having this issue with the latest firmware, please contact us at info@prusa3d.com.

Marina Curzi
Prusa Research

closed 2022-05-17 by Prusa-Support