prusa3d/Prusa-Firmware #774: MK3 3.2 Print Fan Error does not reset bed temperature to 0

rainersc, 2018-05-28:
I currently have a problem with my print fan. i had 2 instances of print fan error (sorted the replacment via chat already)
what i saw was that even though the hotend temperature was reset to 0 , the bed temperature stayed on 80.
i assume it should be turned off as well as there is now possibility to resume the print after that error.

PavelSindler, 2018-05-28:
There is chance to resume print in some cases. You can have some obstacle which blocks your fan and after removing it you can continue printing. That is why fan error behaves as pause print. You can defide if you want to stop print or try to resume.

rainersc, 2018-05-28:
it did nopt offer e any chancce to resume.. ( which would have been nice when it stopped at 90% of a 12h print yesterday).. is there a timeout?
btw. above picture was taen in the morning. the problem occured approx. 10h earlier and the printer was happily heating the room through the night.
so unless there is a possibility to resume, it should be turned off.... at least in case there is a timeout or something

closed 2018-05-30 by PavelSindler