prusa3d/Prusa-Firmware #777: 3.2.1 MK3 Bed Mesh Leveling Failing - Dual Issue

PStewardYUL, 2018-05-29:
Just updated to 3.2.1 and cannot print....
1st Issue which is consistent is after Mesh Leveling on the way to home a crash is detected in the same spot every single time (there is no crash).
2nd Issue which is inconsistent (but consistent location) is during Mesh Leveling it produces an error related to leveling.

Never had either issue prior to the firmware update.

PavelSindler, 2018-05-29:
There is Z crash detection active during mesh bed leveling. This protects steel sheet surface in case that PINDA is too high. 
Try to re-run Z calibration (menu Calibration -> Calibrate Z).  This will ensure that X-axis is parallel to heatbed surface (in case that X-axis is not patallel, false Z crashes can occur). Make sure that there is steal sheet on heatbed during Z calibration and mesh bed leveling.

PStewardYUL, 2018-05-30:
I scrapped my calibration settings and started over. Had to adjust the PINDA (not sure why, it was perfect before the update to the firmware), but it will not complete a calibration of XYZ now (freezes on Y calibration going to step 3). In addition it fails on X and Y selftest in regards to axis length....
Never had issues before...

PStewardYUL, 2018-05-31:
I rolled back the firmware and issue is still there (I brought it back to 3.2.1 now though). The issue seems related to Y axis as it keeps restarting at the Y axis self-test in the wizard and if I move the Y axis via settings too much it will also restart the printer or cause it to home.
Please note: I have had no issues with printer before performing the firmware update.

PStewardYUL, 2018-05-31:
This is resolved. 
Resolution:
Tightened screws on heatbed (at cable entry)
Tightened screws on front pulley on Y axis. 
Reset printer to Factory setting 

The above are not a guaranteed fix, it appears something changed with the firmware that created this issue (as I never had it before). I'm really not sure how this even resolved it, but I really tried everything.

closed 2018-05-31 by PStewardYUL