prusa3d/Prusa-Firmware #778: 3.2.0 Inconsistent mesh bed leveling

martinowren, 2018-05-29:
Been having issues with inconsistent mesh bed leveling, in that sense I have to re-adjust the live adjust Z for every print, even with the same G-Code to get the same first layer.

stahlfabrik, 2018-05-29:
Did you temperature calibrate your Pinda? That is MOST likely your issue.

As a workaround - if you do not want to calibrate - make SURE to start the print always with the same PINDA temperature of 35C - NOT warmer, not colder.

You do that by starting a print with PINDA on ambient temperature. Modify your start gcode to move the PINDA in a heating position. Then use the GCODE "M860 S35" to wait until the Pinda has reached 35C. Then home, then mesh bed leveling and go.

stahlfabrik, 2018-05-29:
https://github.com/prusa3d/Prusa-Firmware/wiki/Manual-temperature-calibration

stahlfabrik, 2018-05-29:
If it all worked great for you with an older firmware, you could create that state by loading the old temperate calibration offsets into EEPROM by "M861 !" and then enable temp cal in the settings in the menu

stahlfabrik, 2018-05-29:
The automatic calibration is not working out good for some users/printers like me/mine. There is no easy solution I am aware of. But the options are not "hard" really. The manual print based calibration is quite easy once you understood it and takes about 2 hours.

martinowren, 2018-05-30:
The temperature calibration takes approximately 5+ Hours for 4/6 stages. Had to cancel it because it took too long. The manual Pinda calibration, if I correct it for PLA, wouldn't it be all screwed if I then changed to ABS with different live adjust Z?

imrichvalach, 2019-01-04:
The automatic temp calibration took <30mins for me and improved the first layer. Although, it is still far from perfect. 
Firmware 3.5.0-1749

closed 2023-09-04 by github-actions[bot]