prusa3d/Prusa-Firmware #792: [Idea] Stop filament oozing at the start

ManuGithubSteam, 2018-05-31:
As we all know the filament does ooze out when we preheat the nozzle before print. This could be counteracted with a retract of the filament in the heater.

As the heater block melts the plastic is is so soft that it should be possible to retract it. And suck it back up. This would minimize the oozing.

What do you think ?

@all 
Can we pull this of with start gcodes ?

Thanks

vaxxi, 2018-05-31:
Check the forums, plenty of solutions there - [one example](https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/no-ooze-mesh-bed-leveling-in-slic3r-pe-1-39-2-t17889.html).

tavisma, 2018-06-02:
Similar to this one (which has a few suggested workarounds)
https://github.com/prusa3d/Slic3r/issues/913

closed 2023-09-04 by github-actions[bot]