prusa3d/Prusa-Firmware #796: File incomplete warning if filename contains an umlaut

mariusoe, 2018-06-01:
Hi,

the latest firmware (also the once before) show the "file incomplete" message (when printing from SD card) if the G-code's filename contains an umlaut/mutated vowel (like: ä, ö, ü) even the G-code file is valid. Renaming or removing all of these characters from the file's name solves the "problem".

bubnikv, 2018-06-02:
by the way, the slic3r pe 1.40.0 renames umlauts in file names to 7bit
ascii when exporting a g code.

NielsPlougman, 2019-01-30:
Also problem with Danisk/ Swidish / Norigan æøå

mumboz, 2020-01-18:
Same problem as in #2130 it sounds like.

Creeper077, 2021-02-10:
Having the same problem, my printer even managed to crash completely when I tried to print this file

mumboz, 2023-08-28:
Still an issue on my MK3S with the latest firmware 3.13.0-6873.

Renaming any file to include e.g. `ü` seems to reproduce it easily.

closed 2023-11-06 by github-actions[bot]