prusa3d/Prusa-Firmware #798: MK2.5 Temperature Calibration never finishes step 6/6

desaster, 2018-06-02:
On my MK2.5, the temperature calibration never finishes step 6/6.

Looking at the temperatures in the debug menu, the PINDA temperature manages to reach around ~56°C, while the calibration wants it to reach 60. At step 6/6, the hotend target temperature is set to 50°C, and the bed temperature at 110°C.

In my case, the problem seems to be that the hotend cooling fan is cooling the PINDA probe just enough to keep it from reaching 60. By forcibly stopping the fan, the temperature calibration soon reaches the end.

Some possible solutions/improvements:

* Keep the hotend target temperature slightly below 50°C to keep the hotend fan from turning on
* Disable the hotend fan by some other means in the code
* Automatically cancel the temperature calibration if it doesn't reach the end after a reasonable time.
* Display the PINDA temperature during the calibration so the user has a better idea of what is going on

PavelSindler, 2018-06-04:
Thank you for your feedback.

geeksville, 2018-08-05:
My new MK3 running 3.3.1 has a similar issue of not getting past step 5/6.  Even if I cover the hotend fan (to keep its turning on from cooling the penda).  So I can't use this feature for now.

cara, 2019-01-31:
Same problem here with MK2.5 and 3.5.1. Still hanging at 6/6

franzbischoff, 2019-04-26:
3.7.0 stuck in 1/6

CCMCAGP, 2019-05-10:
I am also stuck at 1/6 with a stock MK3S (no modification whatsoever). I have been stuck there for 30 minutes already. Firmware is also 3.7.0...

franzbischoff, 2019-05-14:
Firmware 3.7.1 (the latest), Model Mk2.5S

I see in the code that during the first step (1/6):
- My room temperature is 27ºC
- start_temp is set to 35ºC
- the bed temperature is set to (70 + (start_temp - 30)) ==> 75ºC
- while() waits for PINDA temperature to reach 35ºC

But:
My PINDA temperature never gets more than 13ºC. Nozzle temperature reached a maximum of 43ºC.

Is this ok? The 35ºC limit is really for PINDA or should be to Nozzle? Is my PINDA broken?

leptun, 2019-05-14:
13C looks like broken/disconnected thermistor wire of pinda. You should contact support and get it replaced

franzbischoff, 2019-05-14:
Found my problem. My v-cable came wrong

[image]

[image]

mrmoo06, 2019-07-23:
And did swapping the wire over solve this issue?

franzbischoff, 2019-07-23:
Totally

mrmoo06, 2019-07-23:
OK, thanks.  Just checked mine and it is correct but it still just sits at 1/6 with the bed at 75C.....
Who knows, just another thing that doesnt work properly with this thing ##shrug

closed 2023-09-04 by github-actions[bot]