prusa3d/Prusa-Firmware #801: MKS2.5 filament sensor does not detect clogged nozzle

tuxmobil, 2018-06-03:
According to the manual the filament sensor should be able to detect a clogged nozzle.
My experience is it does not. Print completely failed due to filament blockage (extruder clicking) without the sensor noticing it.

FW 3.2.0 MKS2.5

Filament was yellow PLA from "Das Filament"

alientek, 2018-06-04:
They never implemented this feature.

clearbucketLabs, 2018-06-04:
they could detect this via TMC drivers... but that 8 bit CPU is already overloaded.

Area5142, 2018-07-12:
@clearbucketLabs no TMC drives on MK2.5.

Prusa-Support, 2022-05-19:
Thank you for sharing your ideas.

As highlighted above, the MK2.5/S cannot be helped by TMC drivers (missing), but, unfortunately, not even the MK3/S/+ hardware generates sufficient data for detecting a non-moving filament with certainty, or reliably enough.
Our developers are working and experimenting tirelessly on alternative solutions, which may require alternative or additional sensors/hardware. If the research will be successful, the solution will most likely be implemented in future products (probably yet to be invented).

This issue is similar to issue #476, and it will therefore be closed as a duplicate.
Please check said issue and possibly refer to it in case of future discussions.

Michele Moramarco
Prusa Research

closed 2022-05-19 by Prusa-Support