prusa3d/Prusa-Firmware #804: Resuming Print

tbedore, 2018-06-03:
I am running 3.2.1 firmware on my MK3. I heard the gears clicking, so I hit the pause print button, the extruder head went up a bunch, maybe 20 or so mm.  I unloaded the filament and I can guarantee the hot end was definitely jammed, I managed to pull the filament out and then reload the filament back in.  

I hit resume on the print and then it set the extruder heater to 0 and then said extruder heating, and yet it was cooling down, so I went into the menu and hit preheat, once it hit the value it was trying to reach of 205c, it went to start printing again, the z dropped a tiny bit and then just stayed there, so I'm like okay that is weird, I hit resume print again.  And then it just started printing in mid-air.  And this is about 15mm above where the print was paused.  A few seconds later after extruding a tiny bit of filament, it went back to the print.  But this isn't normal.

Panayiotis-git, 2018-06-07:
Just happened to me with a MK2.5 with FW v.3.2.1-576.

I'm printing with XT-CF20 and this filament has the tendency to form big blobs on the nozzle that might drop and ruin the print.
So, I watch it all the time and whenever it grows a blob, I pause the printer, clean the nozzle and then resume.
The above weird behavior happened three out of five times I paused the print to clean the nozzle. 
Even if I select the "Resume print" menu choice again and again, the hot end temperature stays at 0 and the "Resume print" choice is still available.

zerg960, 2018-07-03:
Happened to me as well on a MK3 with FW v.3.2.1-576

MilanNXT, 2018-07-11:
Happened to me as well on MK2 with FW 3.1.0

closed 2023-08-24 by 3d-gussner