prusa3d/Prusa-Firmware #805: Add "// action:pause" and "// action:resume" for filament runout sensor

kortschak, 2018-06-04:
Currently "// action:pause" is used for fan speed error to communicate with OctoPrint, but not for filament sensor error, making filament sensing useless when printing through OctoPrint.

Originally reported as a request to OctoPrint foosel/OctoPrint#2673.

PavelSindler, 2018-06-04:
We are currently working on improvents of crash detection and filament detection when printing using Octoprint.

kortschak, 2018-06-04:
Thanks. It was a disappointing discovery today that it didn't exist - fortunately only on the third layer.

FotoFieber, 2019-06-10:
Just made a print with two filament changes (filament run out) and got no notification in octoprint. Would be nice, if this issue could be solved... :)

marfink, 2019-08-10:
Any news on this feature request?

kevhawk2914, 2020-01-13:
This would be nice to see working!!

kortschak, 2020-01-18:
@DRracer (handle chosen by recentness of commits to this repo) Is there any intention to progress this?

mikulik86, 2021-12-23:
It is nice that since adding basic Action Commands in 3.10.0 the Octoprint is indeed reacting to pause/resume/cancel print from the printer controls. 

**But why still no filament runout!!**

Please, add this functionality so that Octoprint users can be remotely alerted that the printer is actually not printing.

This is what I suggest in the event of filament runout:
```
//action:paused
//action:notification Filament runout detected
```

And after loading the filament and resuming the print ofcourse:
`//action:resumed`

This would be extremely appreciated. Thanks.

kortschak, 2023-08-28:
Still wanted

kortschak, 2023-10-30:
still wanted

3d-gussner, 2023-10-30:
@kortschak Thanks for the reminder. We will look into that. Can't promise when but assigned it to myself.

PR #4751 (2024-08-23): Send host actions on filament runout
- Sends `//action:notification Filament runout detected!` to the host in case of filament runout.
- Adds `action:paused` & `action:resumed` to M600.

Resolves #805
closed 2024-08-23 by 3d-gussner