prusa3d/Prusa-Firmware #807: Potential bug in mesh_bed_calibration.cpp

Arksine, 2018-06-04:
I was browsing your Mesh Leveling and Calibration code looking at potentially implementing something similar in Klipper when I ran upon what I believe are a couple of bugs in the sample_mesh_and_store_reference() function of mesh_bed_calibration.cpp.

On Line 2848:
float zmax = zmax;

I suspect you intended to initialize zmax to zmin here.

On Line 2852:
zmax = min(zmax, mbl.z_values[j][i]);

I suspect you need to use the max() function here.

Tagging @stahlfabrik as recommended by @ff8jake for this issue as well.

stahlfabrik, 2018-06-04:
Oh yeah - that looks fishy. zmax is not initialized at all - or is it? I wonder how that even compiles:-)

I wonder what the results would be for the mesh bed leveling? Is my wave washer adding operations thanks to @ff8jake totally not necessary because the values are wrong? 

ARGH!

stahlfabrik, 2018-06-04:
Well - I am quite sure that is a bug - but it will not influence the mesh bed leveling at all, as it is only used for a sanity check (to stop the process if the bed is more than 3mm off)

But the whole function deserves a review now:-)

Arksine, 2018-06-04:
@stahlfabrik  I'm guessing zmax is initially set to zero in this case, although technically it is being used before its initialized.  The compiler should at least warn.

As you said, this doesn't affect G80, it just affects calibration and warns the user that their bed is WAY too warped.  

The good news is that its an easy fix.

closed 2020-09-22 by DRracer