prusa3d/Prusa-Firmware #815: Set crash # limit for a print

westranm, 2018-06-07:
Using 3.2.0 firmware on my i3 MK3, I was printing a tall object that pushed the extruder to the top of the printer.  The filament became knotted and hung up the extruder, causing 80 recorded collisions and destroying my nearly finished print.

There should be a limit to the number of collisions, preferably selectable by the user.  Once this limit is reached, the printer should pause to allow the operator to clear the issue and continue the print, or cancel the print in case the issue cannot be cleared successfully.

PavelSindler, 2018-06-11:
Thank you for your feedback. We are currently working on it and testing it and it will be released this week.

westranm, 2018-06-11:
Excellent!  Looking forward to it.

closed 2023-08-24 by 3d-gussner