prusa3d/Prusa-Firmware #822: MK2.5 temperature (PINDA) temperature calibraton stays 'stuck' at step (6/6)

Petclaud, 2018-06-08:
I have tried now several times on my MK2.5 to perform a PINDA temperature calibration. It always stays 'stuck' at step 6 from 6. I have performed a full factory reset and also after the initial calibrations, the temperature calibration stays stuck at test 6/6 (the 6th test is at 110 degrees Celcius).
One time I had set the test to run and went to bed and in the morning it was still at step 6/6 (having the bed heated all night at 110 degrees...).

My hypothesis is that during this test at the highest temperature, the bed can just barely hold the 110 degrees Celsius and it might occasionally drop just below the 110 degrees (even if this is not visible on the screen due to rounding) and hence it remains stuck. This is just my guess. 

I have also an MK3 and I can perform the PINDA temperature calibration test without any problem.

DevDocX, 2018-07-04:
I have this same issue on my MK3.  I checked the pinda temp and it was only getting to about 56 out of the desired 60 degrees.  Is this perhaps an issue with the probes temperature sensor, or in the firmware?  I am running MK3 3.3.0 Final.  Please advise as to whether this is a software or hardware issue which I need to contact support over.  Thanks in advance,

Malfate, 2018-07-20:
I am getting the same thing at stage 4/6

rongith, 2018-10-08:
I got the same issue with MK3. In a 22°C room without enclosure, my PINDA barely reached 56°C after 3 hours.
That is because the hotend reach 50°C and then the Noctua fan automatically begin to run and send fresh air to the PINDA. That should more obvious with the R3 revision which is more efficient.

My trick, for the last stage after waiting the bed is 110°C:
- Block the rotation of the Noctua fan with a finger.
- Manually set the nozzle temp to 200°C. The Noctua fan send enough Joules from the hotend to the PINDA to reach the desired 60°C.

Prusa-Support, 2022-05-05:
The PINDA (v2) may not complete the temperature calibration due to a partially damaged thermistor/wire on the probe, or due to fast heat dissipation of the probe preventing it to reach the target temperature. In the latter case, an enclosure or a warmer ambient can help.

This problem has been addressed by the newer Super PINDA, which is temperature independent. The Super PINDA is back-compatible with the printer model MK2.5 (it needs to be enabled via HW Settings on the up-to-date firmware).

This issue will now be closed as obsolete.

Michele Moramarco
Prusa Research

closed 2022-05-05 by Prusa-Support