prusa3d/Prusa-Firmware #823: erroneous movements when stopping a print

stahlfabrik, 2018-06-09:
When I stop a print via menu OFTEN the MK3 keeps printing a little, but in erroneous directions - as if the sign of the "movement vector" gets flipped. It does not happen every time. 
[image]

alientek, 2018-06-14:
Same here +1

PavelSindler, 2018-06-21:
I made pull request which should fix this issue: https://github.com/prusa3d/Prusa-Firmware/pull/860. This erroneous movements could occur also during crash detection before rehoming and during filament detection. It needs more investigation to be sure that all scenarios are covered. We will test it tomorrow. Thank you for your feedback.

stahlfabrik, 2018-10-08:
@PavelSindler the PR went into the firmware? We might close this issue then:-)

closed 2019-03-05 by stahlfabrik