prusa3d/Prusa-Firmware #824: MK2.5 Filament Sensor Does not stop print in time

CoolNamesAllTaken, 2018-06-09:
I recently ran through the end of a roll of filament with the MK2.5 with a filament sensor installed.  The printer paused, but the filament went past the end of the extruder gears and got stuck in the hotend.  I took apart the hotend to remove the filament and will try to end a print with the filament sensor again.  This seems like it is not the intended functionality of the sensor?

clearbucketLabs, 2018-06-10:
could of been the bend in the filament due to being at the end of the spool messed it up.  i don't know if they tested that.  i've had that happen once because of that bend but if you cut it when it's fairly straight it seems to work fine.   however in my case the bend prevented it from going back in the PTFE properly.... not that it was passed the gear.

CoolNamesAllTaken, 2018-06-11:
Interesting--I'm going to try running through the end of a spool again to see if this reoccurs.

Grawp, 2018-06-12:
Duplicate of #523

Prusa-Support, 2022-05-17:
Duplicate of [#525](https://github.com/prusa3d/Prusa-Firmware/issues/525) and  #523 (https://github.com/prusa3d/Prusa-Firmware/issues/523)

Marina Curzi
Prusa Research

closed 2022-05-17 by Prusa-Support