prusa3d/Prusa-Firmware #825: MK2.5 M601 Gcode command not working?

CoolNamesAllTaken, 2018-06-09:
I used to be able to use the M601 gcode command from Octoprint to initiate a long pause (raise hotend, turn off extruder heater, bring bed forward) and the M602 command to resume the print.  These commands have ceased to function properly on the MK2.5.  Are they still implemented?

PavelSindler, 2018-06-13:
Yes, it should still work but these commands are currently for SD card printing only. It works if you print file stored on printer SD and uses Octoprint for monitoring and controlling the printer.

CoolNamesAllTaken, 2019-03-04:
Are M601 and M602 enabled for gcode streaming again?  I still haven't managed to build pause gcode script that works for all extruder temperatures, since it requires finding a temperature to restore to and that depends on the filament that's being printed.

closed 2023-09-04 by github-actions[bot]