prusa3d/Prusa-Firmware #828: M860 S35 Wait Until Pinda threw Octoprint Timeout

danielegger1, 2018-06-11:
I am using the wait until pinda command line:
M860 S35 ; the new code - wait until PINDA is >= 35C
at every print and I am using Octoprint to start my prints.
But there seems to be a timing issue, as octoprint get`s every sixth command in the terminal a timeout as below, which sometimes cause my print to stop at the wait command:

Send: N17 M860 S35*110
Recv: Wait for PINDA target temperature:35
Recv: P:29.8/35
Recv: P:29.9/35
Recv: P:29.8/35
Recv: P:29.7/35
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N18 M105*30
Recv: P:29.8/35
Recv: P:29.9/35
Recv: P:30.0/35
Recv: P:30.1/35
Recv: P:30.1/35
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N19 M105*31
Recv: P:30.1/35
Recv: P:30.3/35
Recv: P:30.3/35
Recv: P:30.2/35
Recv: P:30.2/35
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N20 M105*21
Recv: P:30.2/35
Recv: P:30.4/35
Recv: P:30.4/35
Recv: P:30.4/35
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N21 M105*20
Recv: P:30.4/35
Recv: P:30.5/35
Recv: P:30.7/35
Recv: P:30.7/35
Recv: P:30.7/35
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N22 M105*23
Recv: P:30.9/35
Recv: P:31.0/35
Recv: P:31.1/35
Recv: P:31.0/35
Recv: P:31.1/35
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.

Zawadaki, 2018-06-11:
Try to set "Simulate ok for resend requests" to always

Zaf9670, 2018-06-11:
Do you have the latest firmware compiled that contains the OctoPrint fix? If not you will need what Zawadaki just mentioned in OctoPrint settings I believe.

danielegger1, 2018-06-11:
I have the latest firmware 3.2.1 on the printer and the latest octoprint release version installed.
I will try the simualte ok option tomorrow thanks.

GurliGebis, 2018-06-12:
Try adding the M860 command to the Octoprint list of commands that takes a long time.
It might be because it does not know that it has to wait for it to finish.

Zaf9670, 2018-06-12:
@danielegger1 there are a few commits that have been made for USB printing features and M860 that might improve this whenever 3.2.2+ releases. I thought I specifically saw some OctoPrint commits for sending Wait and OK lines that wasn't in 3.2.1 but now I can't find that specific reference.

PavelSindler, 2018-06-14:
It has been fixed in https://github.com/prusa3d/Prusa-Firmware/pull/744

closed 2018-06-14 by danielegger1