prusa3d/Prusa-Firmware #839: MK3 GPIO Serial Connection PrusaPrint/OctoPrint

thadude3, 2018-06-14:
After upgrading to 3.2.1 firmware we are unable to connect to the MK3.

2018-05-20 22:32:06,260 - Connecting to: /dev/ttyAMA0
2018-05-20 22:32:06,290 - Changing monitoring state from "Offline" to "Opening serial port"
2018-05-20 22:32:06,337 - Connected to: Serial<id=0xa6f1cd70, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2018-05-20 22:32:06,340 - Changing monitoring state from "Opening serial port" to "Connecting"
2018-05-20 22:32:06,392 - Send: N0 M110 N0*125
2018-05-20 22:32:36,423 - There was a timeout while trying to connect to the printer
2018-05-20 22:32:36,464 - Changing monitoring state from "Connecting" to "Offline"
2018-05-20 22:32:36,501 - Connection closed, closing down monitor

see thread:
https://shop.prusa3d.com/forum/user-mods-octoprint-enclosures-nozzles--f65/raspbery-pi-zero-w-suddenly-stopped-connecting-to--t18530.html

firmware version: prusa3d_fw-3.2.1.576-1_75mm_MK3-EINSy10a-E3Dv6full.hex

PavelSindler, 2018-06-14:
Thank you for feedback. We will check it.

Dolnor, 2018-08-09:
Any news on this?
I'm facing exactly the same issues on a fresh PrusaPrint image, trying to connect via GPIO using RPi Zero W. 

Connecting to: /dev/ttyAMA0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0xabb48710, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor

I'm on 3.3.1 firmware however as of today.

thadude3, 2018-08-09:
I was never able to solve this problem. I think my einsy board is defective or something.  I just gave up

Bigfellas, 2018-08-10:
Same issue here. Any updates.

ssabo, 2018-09-24:
Also seeing same problem. Using the PrusaPrint image running Octoprint 1.3.6. Tried updating to 1.3.9 and still seeing same problem.

Disassembled the electronics box and the pins look like they are soldered properly and in all the right places. 

Double checked all the settings in the octoprint image and it appears that /dev/ttyAMA0 is setup correctly and I have rebooted the whole system and printer multiple times trying to get the printer to respond to the connect request. 

```
OctoPrint version : 1.3.9
OctoPi version    : 0.15.0PE
```

```
Connecting to: /dev/ttyAMA0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0xaed8a970, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
```

d1nd141, 2019-01-20:
any news on this?
I printed yesterday something for 8 hours. Everything fine.
Shutdown Raspberry (2b), and powered off the printer.
This morning the connection does not work anymore.

```
2019-01-20 08:04:29,358 - Enabling serial logging
2019-01-20 08:04:34,378 - Connecting to: /dev/ttyACM0
2019-01-20 08:04:34,406 - Changing monitoring state from "Offline" to "Opening serial port"
2019-01-20 08:04:34,434 - Connected to: Serial<id=0x6a38e2f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2019-01-20 08:04:34,441 - Changing monitoring state from "Opening serial port" to "Connecting"
2019-01-20 08:04:34,469 - Send: N0 M110 N0*125
2019-01-20 08:04:46,683 - Recv: echo: 3.5.1-1778
2019-01-20 08:04:46,701 - Recv: echo: Last Updated: Dec 19 2018 20:18:02 | Author: (none, default config)
2019-01-20 08:04:46,705 - Recv: Compiled: Dec 19 2018
2019-01-20 08:04:46,709 - Recv: echo: Free Memory: 2036  PlannerBufferBytes: 1392
2019-01-20 08:04:46,712 - Recv: echo:Hardcoded Default Settings Loaded
2019-01-20 08:04:46,715 - Recv: adc_init
2019-01-20 08:04:47,039 - Recv: CrashDetect ENABLED!
2019-01-20 08:04:47,387 - Recv: PAT9125_RES_X=0
2019-01-20 08:04:47,391 - Recv: PAT9125_RES_Y=240
2019-01-20 08:04:47,394 - Recv: PAT9125_init:1
2019-01-20 08:04:47,397 - Recv: FSensor DISABLED
2019-01-20 08:04:47,400 - Recv:
2019-01-20 08:04:47,662 - Recv: echo:SD card ok
2019-01-20 08:05:16,503 - Recv: MMU not responding - DISABLED
2019-01-20 08:05:46,538 - There was a timeout while trying to connect to the printer
2019-01-20 08:05:46,561 - Changing monitoring state from "Connecting" to "Offline"
2019-01-20 08:05:46,595 - Connection closed, closing down monitor

```

```
Jan 20 08:01:46 octopi kernel: [    2.457247] usb 1-1.5: new full-speed USB device number 5 using dwc_otg
Jan 20 08:01:46 octopi kernel: [    2.604072] usb 1-1.5: New USB device found, idVendor=2c99, idProduct=0002
Jan 20 08:01:46 octopi kernel: [    2.607212] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 20 08:01:46 octopi kernel: [    2.610253] usb 1-1.5: Product: Original Prusa i3 MK3
Jan 20 08:01:46 octopi kernel: [    2.613193] usb 1-1.5: Manufacturer: Prusa Research (prusa3d.com)
Jan 20 08:01:46 octopi kernel: [    2.616110] usb 1-1.5: SerialNumber: xxxxxxxxx
Jan 20 08:01:46 octopi kernel: [   19.015525] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Jan 20 08:01:46 octopi kernel: [   19.458912] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
Jan 20 08:01:46 octopi kernel: [   19.840771] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device

```

OctoPrint version : 1.3.10
OctoPi version    : 0.15.0PE

Firmware 3.5.1

ssabo, 2019-01-20:
Prusa sent me a new Einsey board and that fixed my problem. @d1nd141 your problem looks slightly different since you seem to get a response from the printer but the connection is crashing in startup (I think?)

3d-gussner, 2020-09-22:
@thadude3 Is this still an issue with newest firmware? If not please close this issue.
[stale]

thadude3, 2020-09-22:
Still an issue, its never worked.

closed 2023-09-04 by github-actions[bot]