prusa3d/Prusa-Firmware #854: feature request: multiple z-adjust slots

kortschak, 2018-06-20:
Given that the MK3 can have more than one print surface and that different print surfaces may have different z-adjusts, it would be nice to have more than one slot to store the z-adjust (at least two - one for smooth PEI and one for powder coat). With this feature changing print sheets would then involve just selecting the z-adjust from a settings menu, rather than having to do a live adjust at the start of the print.

OutlawAndy, 2018-08-23:
Bump!

Anything happening here?  Is this feature being considered?

m-cas, 2019-03-14:
I guess that this would probably need to be managed in a more comprehensive way as the z-value will change for each of the combinations between bed, filament and layer height.
Would be nice to be able to store such a matrix (that I hold in excel and set manually when changing any of the above factors).
Something for slic3r? Add a tab with z-adjust values and send it to the printer via GCODE?

kortschak, 2023-08-28:
Still wanted

3d-gussner, 2023-08-29:
@kortschak There are multiple sheets in Settings -> HW Setup -> Sheets with the values. Also the gcode `M850` allows you to read and set the sheet name `M850 - Set sheet data S[id] Z[offset] L[label] B[bed_temp] P[PINDA_TEMP]`

Example to read first sheet: `M850 S0`
Result on my printer: `Sheet 0 Z-0.8000 R-320 LSmooth1 B0 P39`

3d-gussner, 2023-08-29:
Please consider to close this issue.

closed 2023-11-06 by github-actions[bot]