prusa3d/Prusa-Firmware #856: Heatbed Max-Temp limit?

zawadzkip, 2018-06-21:
Working with someone in a facebook group who is new to printing, and they were misreading some temperature settings for a filament they were using and set the max-temperature of their heat bed to 260C.  Maybe setting a hard-limit for 100c +/- 10C would be sufficient? I haven't found many filaments myself that need heated beds above 70C or so. Any suggestions?

Maybe the firmware isn't the best place to put a max-temp limit, but I think it is a good place simply because it's the "last resort" catch-all for errors like this that can potentially cause severe damage.

https://www.facebook.com/groups/892917067538312/permalink/1046354168861267/?comment_id=1046370248859659&reply_comment_id=1046383515524999&notif_id=1529596270677369&notif_t=group_comment_reply

stahlfabrik, 2018-06-21:
I see a #define BED_MAXTEMP 125 in the EINSY variant file. Does it not work as a limit? How hot did it go?

zawadzkip, 2018-06-21:
@stahlfabrik The user set their printed to 260C, but we did actually notice it when the bed was heating at 45C. Is it possible to have the firmware check before it reaches that top-limit? When is it loading the heat bed temperature?

cnlson, 2018-06-29:
@zawadzkip not sure if you are talking about a prusa printer or not. as stahlfabrik says it is limited in the firmware. I print polycarbonate from time to time and it requires 120C so the 125 limit is fine for me.

zawadzkip, 2018-06-29:
@cnlson I am talking about the Prusa printer, the person was using an MK3 and set his printer to 260C on the heatbed. When you print polycarbonate, does it warn you when it reaches it's maximum temperature or does the firmware detect the gcode trying to set the heatbed to a temperature over the limit and prevent it then?

It's a nice to have, but my idea, sorry if I'm not explaining it well, would be to have the firmware check the limits when it's reading the Gcode, as opposed to letting the heatbed/nozzle heat to their max temperatures and then fail.

cnlson, 2018-06-30:
@zawadzkip @MarekC3D @josefprusa Ok i decided to try it just for kicks and you are correct, the firmware heatbed limit does not seem to function If the setting is in the gcode. 

[image]

zawadzkip, 2018-07-03:
@cnlson that's good to see that it fails and that the feature is working when the bed hits max temp, but it doesn't make sense to have it heat all the way to that max temp before failing, it should just read the Gcode, and compare that value with the firmware value and then throw errors accordingly.

closed 2023-08-24 by 3d-gussner