prusa3d/Prusa-Firmware #865: Date Sort does not work with FlashAir SD card

Tinker999, 2018-06-23:
Date Sort does not work with FlashAir SD card. When a file is save over the WIFI to the FlashAir the order appears to be random. I believe the reason might be that the "Date Created" field  appear blank when viewed from Windows 10. The "date Modified" and "date last save" fields are OK. If the sort is sorting "date created" this would explain it and a easy fix would be to use   "date Modified" or "date last save". Thanks you in advance.

cnlson, 2018-06-30:
i've seen this as well. makes it hard to find the latest uploaded file

bobstro, 2018-07-10:
IME, the FlashAir card sets all dates the same when accessed via the network. When you pull the card and stick it in your computer, you'll see all of the files set to the same date. I'm seeing a lot of dates set to December 31st, 1969. If you read/write to that same card as a normal SD card, everything works as expected. This is due to how the FlashAir card is writing the files.

Tinker999, 2018-07-15:
I agree “date created” is bad but "date Modified" and "date last save" fields are OK. So if the sort would change fields it would work.  For now I’ve given up on date and use alphabetical.

cnlson, 2018-07-17:
@Jonnie-Cache No, if you look again pavel fixed the original issue. there was a pull, what you see is a separate issue which was entered _after_ the fix for the first issue. it is documented better here and is not a dup.

DRracer, 2020-09-22:
Sorting files based on "time" may look simple, but different operating systems are very creative in updating some/all of the 3 different time stamps on FAT32. The best solution we could come up with is using the most recent time stamp of both creation and modification time. This is the current approach since MK3/S FW 3.9.0 and the Original Prusa MINI as well.
Thus I'm closing this aged issue...

closed 2020-09-22 by DRracer