prusa3d/Prusa-Firmware #876: CoreXY

Patrik-svobodik, 2018-06-26:
Hi, this is not an issue, it is question. Is it possible to use einsy rambo, pinda2 and filament sensor with prusa fw on core xy printer? Can prusa fw handle it?

Patrik-svobodik, 2018-06-26:
My friend sold me a cheap mk3 but I am not satisfied with cartesian

stsfin, 2018-06-26:
Its a Marlin, so you can modify it the way you want, Prusa probably wont since they dont sell a CoreXY.

clearbucketLabs, 2018-06-28:
it's not marlin, all that stuff is stripped out.   it's way behind marlin so use marlin on the board or buy a duet Wifi.  it's a better board and you can then get the expansion to use the e3d toolhead changer which IMO is better than MMU anyway... no purge.

Merdeka, 2018-08-02:
Have you tried the following:
- Get the MK3 Source code from the MK3 Branch at https://github.com/prusa3d/Prusa-Firmware
- Add the Rambo Board in the Arduino IDE (https://raw.githubusercontent.com/ultim ... index.json)
- uncomment #define COREXY on line 212 in Configuration.h
- comment #error BABYSTEPPING not implemented for COREXY yet. on line 274 in Configuration_adv.h

It should compile now and you could start from there.
Maybe port in the BABYSTEPPING stuff for CoreXY from Marlin?

Keep us informed of your progress/troubles.

Cheers,
Timo

Patrik-svobodik, 2018-08-02:
Hi, I gave up and got really good deal on duet wifi from my friend, so I am selling the einsy rambo, you can get it cheap if you want, just email me @ svob.pata@gmail.com

closed 2023-09-04 by github-actions[bot]