prusa3d/Prusa-Firmware #879: [another request] After a finished print add a new menu entry "repeat last job"

RacingHell, 2018-06-26:
Hi,
I have seen this feature recently on some printer review (forgot which one it was). I think this repeat-option might be very uselful for many users if going for a couple or more copies of the same object.

Sammy

Zaf9670, 2018-06-26:
I wonder if this could be implemented with a custom GCODE. End of script waits for confirmation that the bed is clear and repeats from the start of the GCODE. Or you can hit no if you don't want to repeat.

Not sure how much time it would save for a typical user but I don't think it would be hard to implement.

BrienKing, 2019-08-12:
I would like to see this added as well.  It's a huge pain when I'm printing 10 of something and it's deep down the list of objects to print.  This would save me A LOT of time.

closed 2023-09-04 by github-actions[bot]